scottbez1/splitflap #8: cards

raoul170, 2016-02-11:
the widest point on the "tab" is 1.5 mm correct?  Can it be square?  Can the "hight" (15mm) be lower?
[image]

scottbez1, 2016-02-11:
The thickness of the tab is mostly determined by `flap_hole_radius` in `splitflap.scad`, which is [currently 1.2mm](https://github.com/scottbez1/splitflap/blob/7ef2539ad5551fa2707af5be54aa41065ebd93ef/3d/splitflap.scad#L94).

That gives a theoretical max tab thickness of 2.4mm, but of course it needs to be smaller because the cards have some thickness and you want them to be able to rotate easily within the 2.4mm hole. In practice the cards that I've made have a tab thickness that measures somewhere between 1.2 and 1.6mm.

Square edges would work just fine, and in fact I'd prefer those but couldn't come up with a cheap/easy way to fabricate cards with square edges - would love to hear if you have suggestions there! Note that after using the slot punch on the cards I used diagonal cutters to snip off the sharp points left over to make the holes slightly more square - see the last 2 photos here: https://github.com/scottbez1/splitflap/wiki/Cut-flaps

The necessary height of the cutout is slightly tricky. It's important that it's a certain minimum size to allow the flap to swing all the way tangent to the spool (giving the flap a full 180 degrees of rotation):
[image]

That value is a little annoying to calculate by hand, so `splitflap.scad` does a [rough calculation based on the input parameters](https://github.com/scottbez1/splitflap/blob/c2433e25229c66a62a86a215dc629cc6730d6cef/3d/splitflap.scad#L120) and prints it out during rendering:
```
ECHO: flap_notch = 10.4717
```
That's likely going to be on the lower bound of what's necessary though; empirically I'd guess anything less than 12mm might start causing issues, and the 15mm from the punch gives plenty of margin.

raoul170, 2016-02-11:
I am concidering to produce 500 sets (20k flaps) with card manufacture. Printed white letters on black "card"   They need to make some stamp mould for it.

Any Ideaa what font used on the Vinyl stickers?  Cannot find such narrow/high font for now anywhere. "Gothic black" seems not correct.   The "robo" font in seems also too wide.

scottbez1, 2016-02-11:
500 sets?!? Wow, that's some impressive volume! At that scale I'm not sure
how helpful my advice will be considering I've only built 2 prototype units
so far, but I'd love to hear how your project goes!

Also depending on your plans and whether you're looking to build full units
based off the designs here, I feel like I should caution you that
everything in this repo is very much under active development -- e.g. in
order to get the IR reflectance sensor to work well, it's very likely I'll
need to add another laser cut hole to one of the spool pieces soon, but
that's not currently reflected in the 3d model.

As for the font, there is a small discussion of the "correct" typeface
originally used on Solari boards here: (sorry for the archive link, I can't
seem to find the info on their current site any more)
http://web.archive.org/web/20150802041313/http://beachlab.org/flip-flap

I favored low cost in small quantities over authenticity though, and just
went with whatever font I could find with appropriately sized stickers :-P

closed 2016-02-17 by scottbez1