scottbez1/splitflap #12: Consider PCB variant using Arduino Micro?

scottbez1, 2016-05-01:
Unfortunately the current PCB design is a little tricky for a beginner hobbyist to assemble due to the surface mount TQFP ATmega32U4 and micro USB connector which take some practice and patience to solder (the resistor pack is also tricky to solder, but that could easily be replaced by individual 0805s).

It might be interesting to design a variant that uses an Arduino Micro instead, with only through-hole parts. It would be more expensive but would make the electronics more approachable (Micro costs ~$25, minus the ~$8 you save by not purchasing the chip and peripherals standalone: resistors, capacitors, usb port, crystal, etc). I suspect it'd still be possible to fit within the 50x50mm desired size for low cost PCB production.

scottbez1, 2016-06-13:
It's also tempting to consider an Arduino Uno shield, since Uno's are ubiquitous, but with the current driver design an Uno can only support 3 modules: each module needs 4 stepper pins + 1 sensor pin, but the Uno only has 18 IO pins available since Rx/Tx are needed for USB serial.

So there are a number of different approaches to consider with different tradeoffs:

| Option | MCU | Driver | Supported Modules | Requires SMD soldering | Notes |
| --- | --- | --- | --- | --- | --- |
| 1 | ATMega32U4** | 3x ULN2003A | 4 | Yes | Original design |
| 2 | Arduino Uno (shield) | 2x ULN2003A | 3* | No | Limited to 3 modules by IO pins |
| 3 | Arduino Uno (shield) | 2-4x MIC5842 | 4-8?* | No | Need to research MIC5841 more. |
| 4 | Arduino Micro (plug in) | 3x ULN2003A | 4* | No | Arduino Micro board isn't common, easily confused with Pro Micro, etc |
| 5 | Arduino Uno (shield) | 1-2x MC33996 | 4-8?* | Yes | Potentially enables driving more modules, but need to research MC33996 more |

\* Supported module count is a guess; might be lower depending on PCB size (50x50mm) constraint.
\*\* It looks like this could be downgraded to a U2 series 32-pin AVR for potential cost savings and smaller footprint; using PB0-7, PC4-7, PD0-7, with PC2 spare (e.g. single debug led).

One other related area for exploration is the possibility of stackable shields or expansion boards to support additional modules. With serial/shift-register outputs like the MIC5842 or MC33996, and potentially shift-register inputs for sensors, it seems theoretically possible to support more modules from a single microcontroller, though I haven't given too much thought to the serial data rate needed to drive the steppers.

scottbez1, 2017-03-24:
For prototyping I ended up ordering the MIC5891 rather than the MIC5842, partly by accident, but it's also ~$0.80 cheaper in single quantities at DigiKey -- $2.04 each in a DIP package. The MIC5891 is a high-side driver as opposed to the MIC5842 (or existing ULN2003A) but either should work fine as far as I know. 

Updating the stepper driver code to use SPI for shifting out data (as opposed to direct port manipulation) was pretty straightforward and seemed to work just as well.

Overall this seems like the way forward for the electronics: the board can now just be an Arduino shield that's entirely through-hole soldering, which should make it way more approachable for hobbyists than the SMT soldering and DFU programming needed with the current design. Using a shift-register driver also allows the driver boards/shields to be easily chained to drive even more splitflap modules from a single microcontroller.

And assuming we also use shift-registers for the sensor inputs, the "shield" should only need to connect to the ICSP header on the Arduino which connect to the high speed SPI pins regardless of specific Arduino model.

scottbez1, 2017-09-30:
Work is ongoing in the [shift_register branch](https://github.com/scottbez1/splitflap/tree/shift_register) on this new shift-register based design.

Arduino code has been updated to support driving chained MIC5842 shift register outputs and uses SN74HC165 shift registers for sensor inputs.

The plan is for a single Arduino Uno to be able to drive up to 12 character modules, via 3 chained electronics boards. Each PCB would have 2 MIC5842 output registers for driving 4 stepper motors and 1 SN74HC165 for reading the sensor inputs, and ideally will be entirely through-hole soldering for ease of construction.

The new design has been tested with 3 modules, with prototype breadboard electronics. Work on an updated PCB design will begin soon.

closed 2018-02-08 by scottbez1

whitebird, 2019-02-01:
One method you can use to simplify design is to allow place for a generic Arduino Micro board. That way you don't have to solder the micro usb port and atmega32u4.