scottbez1/splitflap #14: incorrect rod size

fpagliar, 2016-12-01:
I see the materials list includes:
 5mmx100mm rod
but judging on the pieces I got from Ponoko, it seems like it should be 4.5mmx100mm. 
Is this the case?

scottbez1, 2016-12-01:
Hey, sorry I need to improve the documentation a bit! The latest design on the tip of master is actually an experimental one, that replaces the 5mm rod with 2 M4 bolts facing inward from the sides, which is different than what's shown in the wiki. I haven't actually tested this design yet, but it sounds like you've ordered this latest design from Ponoko, so let's see if it works!

The idea is that each of the sides have an M4 bolt facing inward, with a nut on the inside to hold it in place. (Those nuts takes the place of the bushings that used to be there as mentioned in #15). So I think you want to attach those bolts to the left/right sides, and then put all but one side of the enclosure together, slide the spool onto the bolt and hold it in place while you slide on the last enclosure side, lining up its bolt with the hole on the other side of the spool as you do so.

Hopefully that makes sense? Wish I had some photos I could share, but maybe you could take some to help improve the wiki if this design works out as planned?

Thanks! Let me know if you run into any problems!

fpagliar, 2016-12-02:
Great, I'll try it out, thanks!

fpagliar, 2016-12-05:
Great! It works correctly. 
Thanks!

closed 2016-12-05 by fpagliar

scottbez1, 2016-12-08:
Awesome! Glad to hear it. I'd love to hear how the rest of your build turns out!