scottbez1/splitflap #16: Numbers only

samer-riad, 2017-02-03:
Hey Scott,
I was wondering if I wanted to make it numbers based only. No characters. How would this affect the code and the size of the wheel hosting the cards and the entire mechanism? 
Cheers,
Sam

scottbez1, 2017-03-07:
Hey Sam, sorry for taking sooooo long to respond! I haven't looked too much into making a number-only display, but here are some thoughts from playing around with it for a few minutes:

You can easily decrease the number of flaps by adjusting the [num_flaps variable](https://github.com/scottbez1/splitflap/blob/f1d1edbba32aee4835e0c66f7c48eed6c3641932/3d/splitflap.scad#L95), but this reveals some other issues:

[image]

(In this example I decreased the number of flaps to 10, though you would maybe want to include at least one more so there can be an empty flap in addition to numbers 0-9)

The things you see sticking out of the center there are the "spool struts" which make sure that the disks on either side of the spool turn at the same speed. On the 40 flap design, those struts fit into rectangular holes in the disks and hold the spool in alignment:

[image]

But when the spool disks are much smaller for just 10 flaps, there isn't enough room for those struts in the center of the spool any more, so some modification would probably need to be made, such as replacing the fixed axle with a rotating axle, where each side of the spool is glued or attached to the axle to keep them aligned. Depending on materials and desired longevity, you would maybe want to add bearings on the enclosure to allow the axle to spin with less friction.

If we set aside the struts and look at the other side, the gear appears to be sticking out of the front:
[image]

We can change the size of that gear by adjusting the [spool_teeth variable](https://github.com/scottbez1/splitflap/blob/f1d1edbba32aee4835e0c66f7c48eed6c3641932/3d/splitflap.scad#L105) and shrink it down to something like 10 teeth:
[image]

However, now the motor has a gear with 40 teeth and the spool has 10 teeth, meaning the spool will spin 4 times for every revolution of the motor! The motor may not have enough torque for that, and such a high gear ratio also reduces the precision available from the stepper motor. So it would probably make sense to also decrease the [motor_teeth variable](https://github.com/scottbez1/splitflap/blob/f1d1edbba32aee4835e0c66f7c48eed6c3641932/3d/splitflap.scad#L103).

This of course introduces a new minor problem - with the smaller gears, the motor ends up much closer to the flaps. This is fine for a single display, but if you were to place 2 next to each other, the flaps of one display would hit the motor from the adjacent display:
[image]

One solution to that is to increase the size of the idler gear, which doesn't affect the gear ratio at all:
[image]

At this point the design is getting pretty close to something that could work, though there are a few more details that would need to be worked out (home position sensor/pcb mounting holes, bushing sizes, etc). I'd encourage you to check out the OpenSCAD code and play around with changing different variables and seeing what happens! I'm sure there are tons of variations and possibilities I haven't even considered yet, and I'd love to hear what you come up with.

DurandA, 2017-06-14:
Thank you for the *very* detailed answer. I'm also interested in building an Internet-connected counter out of it!

bobc, 2018-02-21:
I'm also interested in a numbers only display for a clock. 

One workaround for the mechanical issues might be to repeat each digit several times, so 4 times 10 gives 40. Then you need to adjust the control circuit to advance 4 flips to get to the next digit.

dwightmulcahy, 2019-05-10:
was reading that another option is to do 20 flaps instead, 2 sets of the digits+blank.  was hoping that by doing numbers only that this would help reduce some complexity.  Has anyone tried making a number only one?

tree183, 2019-08-02:
I’ve also been thinking about a number-based one.
I have a deck of cards which are beautiful and using them in a flip clock would be a great way to show them off.
Alternatively to create a single display for all 56 cards (jokers and two extras) and have it rotate randomly would be cool. Any thoughts would be appreciated!
The height and depth of the unit would need to be increased and I think the cards would need to sit in a carriage because I wouldn’t want to cut them or stick them, and changing the deck would be a cool feature!

whitebird, 2019-11-14:
I'm currently working on a fork where i'm tinkering with the variables of the openscad render file to enable a numbers only display.

A few things need to be adjusted to build a display with fewer flaps.
- `num_flaps`- The amount of flaps. I've set this to 11 so i have all digits and a blank one.
- `flap_gap` - The distance between flaps. This needs to increase to avoid the issue where the spool is too small and the flap holes overlap with the strut:
[image]
- `spool_strut_tab_width` and `spool_strut_tab_outset` - The size of the inner strut and its tab. Needs to be smaller.
- `magnet_hole_offset` - Ensure that the magnet doesn't overlap with the motor, strut and flap holes.
- [Line 978](https://github.com/scottbez1/splitflap/blob/6a04fa0cf2ef832a47301894900d9eb7ea5edac1/3d/splitflap.scad#L978) rotates the motor. This needs to change so the mounting holes of the motor are within bounds. You also need to check if it overlaps with the hall effect PCB.
- Module `motor_mount()` on [line 432](https://github.com/scottbez1/splitflap/blob/6a04fa0cf2ef832a47301894900d9eb7ea5edac1/3d/splitflap.scad#L432) needs to be rewritten to compensate for the new motor position.

It's also a possibility to move the magnet and hall effect sensor to the other side.

This is the intermediate result that I have, but I have to rewrite `motor_mount()` still.
[image]
[image]
[image]

My fork is over at https://github.com/whitebird/splitflap
@scottbez1, do you think this would work?

thegamebegins25, 2025-01-07:
@scottbez1 Any updates on this with the upcoming release of v2? I'm also attempting to create a numbers-only display.