scottbez1/splitflap #18: More cards

WhiskeyTuesday, 2018-02-27:
Hi Scott, I read through the [10-card numbers only issue](#16) and it got me to thinking about increasing the number of cards so that a user could have a number of reserved cards for symbols, colors, etc. It seems to me that the limitation is the physical size of the spool, and that the whole unit would have to be upsized to accommodate this. I may do a fork for a splitflapXL type project and mess around with the SCAD files a bit but I'm wondering if you have any thoughts or tips before I get into it.

qwertyaardvark, 2018-05-05:
I also second the idea of making a spool that could hold more cards. Aside from making more characters, it could also be a chance to move to a direct drive design and to slim up the gap between characters. By making the spool large enough, the motor could fit inside the spool. The spool's "bearings" would the motor output shaft and the motor support structure. Without gears to the side (or motor to the side as in the gearless2 design), the whole design would get slimmer. 

That said, it seems that the badge punch notch width drove the size of the spool. Looking at the scad rendering, the upper part of the notch is already very close to the spool's edge. If the spool were upsized, you would likely need to find a badge punch with a wider notch or make another tool to double notch it. 

Ideally, there would be no material above the 1.5mm tab. That way, any size spool could be accommodated and no flap would look out of proportion to its spool size. The front bezels would be slimmed down to the new card width and would have notches sized appropriately to let the front part of the spool stick out. But easier said said than done... at this point we're talking tooling and die cut cards.

Some silver lining: I am using these modules for a project at work, so the tooling costs would be paid for by the company. Maybe one day I can provide pre-cut and pre-printed cards for this project ;)

scottbez1, 2018-05-05:
> By making the spool large enough, the motor could fit inside the spool. The spool's "bearings" would the motor output shaft and the motor support structure. Without gears to the side (or motor to the side as in the gearless2 design), the whole design would get slimmer.

Interesting idea. I played around with a somewhat similar idea a while back, though it somewhere between what you've suggested and the gearless2 design - the motor is direct drive but still sticks out the side, but could "nest" inside the spool of the adjacent module to get them close together:

[image]

This was on the "gearless" branch, which is pretty outdated at this point, but was an interesting experiment. It does require more flaps (minimum of 52 or so, based on the size of the motor).

Renee9798, 2019-04-04:
Hi Scott,

As a quick way of increasing the number of cards each module can support, what should I look out for if I were to increase the `num_flaps` variable in `splitflap/3d/splitflap.scad` and then regenerate the laser files?

I saw your post on `imgur.com` which shows that wall length can be an issue if this is taken to the extreme of 100 flaps. What other issues could I encounter? As an example, the following set of characters requires 56 flaps, and I'd love to have the flexibility of all these symbols in my project.

`A - Z`, `0 - 9`, `(space) , . ? / : ' ! @ # $ % ^ & * ( ) - + =`