scottbez1/splitflap #19: double lines in SVG

Neon22, 2018-03-10:
Its possible to save costs in Ponoko and other laser cutters by removing unneeded lines.
If two lines overlap then one can be removed.

I use the inkscape svg templates from Ponoko which scale well (rather than Illustrator tags).
In any case there are a number of lines that are redundant and only one needs to be cut.
Not sure best way to detect and remove from SVG or on export from openSCAD.
I suspect a separate process to remove manually identified lines might be simplest.

scottbez1, 2018-03-10:
Yeah, great point! This note is buried in the README:

Once the combined.svg file is generated, you'll want to manually remove a couple redundant cut lines that are shared by multiple adjacent pieces, to save time/cost when cutting. In Inkscape, select the "Edit paths by nodes" tool and select an edge to delete - the endpoints should turn blue. Then click "Delete segment between two non-endpoint nodes", and repeat this for all other redundant cut lines.

But it would certainly be nicer to do this automatically as part of the export. Not sure the best way, but as you say, it might be easiest to just post-process them somehow.

scottbez1, 2018-05-03:
I've just pushed a commit on a development branch that automatically removes overlapping lines. It's fairly basic, so it only handles overlapping lines where one line fully encompasses the other, but that should cover the common cases. I've also updated the rendering code to highlight the removed lines in red in the raster preview, which looks like this: 
[image]

qwertyaardvark, 2018-05-09:
Aside from Ponoko not willing to even run a drawing with overlapping lines, maximizing the amount of shared lines goes a long way to saving on laser cutting time costs. The attached two layouts of the new gear housing demonstrates this. It started as a challenge to fit all the pieces without taking material out of the main housing panels.

P3 Laser Cost, both fit 8 sets of the housing:
Taking the SVG as shown above and copied 7 more times, butting copies up against each other: $95.57
Placing and fitting all pieces to touch each other as much as possible: $65.87, savings of 31%

[Gearless Housing - Tale of Two Layouts.zip](https://github.com/scottbez1/splitflap/files/1989681/Gearless.Housing.-.Tale.of.Two.Layouts.zip)

Neon22, 2018-05-10:
significant. well done

scottbez1, 2018-09-04:
Going through older issues; just wanted to mention here that basic support for panelizing the design was added back in 4a5fcf6 which allows you to export the rendered 2d design with multiple copies aligned to optimize some laser cut lines. You can do this by passing `--panelize 4` for example when running `generate_2d.py`. Once the panelized design is exported the post-processing step I mentioned above is run, which will eliminate many cases of overlapping straight lines to reduce laser cutting cost. Because of the automated nature of this, it isn't necessarily 100% optimal, but it should handle a good chunk of basic optimization without requiring manual effort.

I'm going to close this issue for now, but feel free to reopen if there are still some specific laser-cutting optimizations that you believe can/should be automatically applied, which weren't covered by the redundant line removal in 2fc1456 or panelization in 4a5fcf6.

closed 2018-09-04 by scottbez1