scottbez1/splitflap #20: Clarification - Motor and Serial

mavbrace, 2018-04-22:
This is a question about sending a message over serial to the units, hoping to get some clarification. Mostly code-related.

We have assembled everything we need to control one node/character from the Arduino. Upon powering up the unit, the purple lights flash and return to orange, giving us the message: 'gave up looking for home’. I've tested our sensor on a separate device and know that it functions. We can also get our single motor to rotate when we enter a '@' into serial. I assume the problem is that the sensor is not assembled behind the gear yet.

The question is, is that assumption correct? Also, do we need a minimum amount of nodes/motors attached for it to function properly?
Also, more importantly, how is the serial communication intended to be used? (i.e. what order did you intend the characters to be sent, eg. the '=').

Really appreciate your work in making a Split Flap Display!

scottbez1, 2018-05-03:
The easiest way to debug a new module is to start with `HOME_CALIBRATION_ENABLED` set to `false` in splitflap_module.h (https://github.com/scottbez1/splitflap/blob/20d789cf1c1a1e8ac40509949c4275f6cd6ed7e2/arduino/splitflap/splitflap_module.h#L30) - this will cause the module to operate in open-loop mode and ignore all sensor input. So if you command the module to move from A->B the stepper motor will still turn the appropriate number of degrees, but the module won't auto-calibrate to the home position or be able to detect if it's out of sync.

Another thing to note: make sure you set `NUM_MODULES` to the appropriate number (https://github.com/scottbez1/splitflap/blob/20d789cf1c1a1e8ac40509949c4275f6cd6ed7e2/arduino/splitflap/splitflap.ino#L49). This is particularly important when `HOME_CALIBRATION_ENABLED` is set to `true` - if it's higher than the number of modules you actually have connected, the Arduino will attempt to read additional bytes from the shift register, but its input may be floating, causing erratic behavior as the code thinks that some "ghost" modules are actually hitting the home position unexpectedly at random times.

To test the sensor, you may want to attach a voltmeter or oscilloscope to the sensor pin while everything is connected and then manually rotate the spool so the hole in the gear passes the sensor - make sure you see a good voltage swing on the sensor pin. If it doesn't swing much you may need to increase the contrast between the side of the gear and the hole - try making the hole darker by coloring the walls with black marker if necessary and the side of the gear lighter using paint, white-out, or a white sticker.

The serial protocol is still kind of a mess at the moment, but currently you would send `=` followed by a letter for each module, followed by a newline. For example, `=abc\n` would command the first 3 modules to `a`, `b`, and `c`, respectively. Any additional modules beyond those specific will remain unchanged, i.e. if you were to follow that command with `=x\n`, the first module would flip to `x` and the other two modules would remain on `b` and `c`. 

`@` will re-calibrate all modules to the home position, so can be useful when first testing the sensors - you can use this to make sure the home position is reliably detected.

scottbez1, 2018-05-05:
Closing this - let me know if you have any other questions!

closed 2018-05-05 by scottbez1