scottbez1/splitflap #24: Split Flap Card Design and Mass Production

qwertyaardvark, 2018-05-05:
[Github Splitflap Characters.pdf](https://github.com/scottbez1/splitflap/files/1976770/Github.Splitflap.Characters.pdf)

This is more of a mental note for when I would make the jump to mass produce the split flaps.

I am intending on making a good number of these modules at work and is an understatement to say that I am not looking forward to making the cards. I have inquired at some vendors. I am waiting to hear back on all quotes, but ballpark seems to be ~$30/set @ 100 sets, ~$15/set @ 300 sets. As these modules are still experimental, I don't think it would make sense to buy these right now. That being said, I need to have these made sometime this year.

The main thing on my mind are these:
-Number of characters in a set (10, 40, 48, 52, 60 etc)
-Overall card shape (notch in the side like current design, or rectangle with just 1.5mm tabs sticking out)
-Overall card dimensions (size of bezel opening, spacing between spools, tab size, use of 3.2mm MDF)

Attempting to guess the future, I presume the spirit of being able to make these cards at home with only a badge punch and cutting jigs will mean the overall card dimensions stay the same. As far as guessing the number of characters in a set and card shape I see two paths:

1) I would opt to shift to a rectangle with tab sticking out design so that I wouldn't have to worry about spool sizes affecting the notch size and redesigning the front bezel (fit opening to new card width, add notch for spool to stick through) which wouldn't be a big deal. Would increase the perceived gap between characters.

2) Keep the current design shape and design the notch for a 60-character spool. Would maintain the current perceived gap between characters.

qwertyaardvark, 2018-05-16:
Running different number flaps and looking at the echo for split flap notch width yielded the following:

40 flaps: 10.4717...... 48 flaps: 11.4208...... 52 flaps: 11.867...... 60 flaps: 12.7124

Seeing that the current 15mm notch is still a safe amount away from the 12.7124, I am going to stick with the 15mm notch as it is designed now. Currently I am looking to having a die made, and finalizing what special characters I need. Right now, the special characters will be period, dash, and colon. The die will be made for a 40 character set. There are thoughts of using the 'zero' for both 'zero' and 'o', and more drastically, the 'one' for both 'one' and 'i' (would have to pick a font where 'i' and '1' look the same) to allow for more special characters, but for now I will stick with 26 letters, 10 numbers, 3 special characters and 1 blank.

Another consideration I've looked into is the mil thickness of the material (rigid vinyl). I will receive proofs for 20 and 30 mil rigid vinyl. Not only is the 20 mil (a little bit) cheaper, but with less material, the torque required by the motor to overcome the spool inertia will be reduced. Most importantly, I hope the sound of the flaps hitting each other isn't too muted! I did get 15 mil and 10 mil samples and I felt they were too wimpy for the sound they would provide, but they would be cheaper and further reduce the spool inertia the motor would have to overcome. Perhaps this would allow the use of the weaker 5V motors and still be reliable? (I one day hope to have these split flaps solar powered... would be nice to only need one voltage supply...) On the bright side, the die can be used for all the thicknesses mentioned above, so I can choose freely between the thicknesses as desired.

@scottbez1 if you are interested, I can send you some flap sets (with or without print) to play with :)

scottbez1, 2018-06-23:
Have you made progress on this? I'd love some flaps to play with if you did end up getting some! Right now I'm gathering quotes from suppliers for die cut blank cards as well, so I'm curious if you've had any luck already.

qwertyaardvark, 2018-06-26:
I have settled on a company to work with, and I am hoping to receive the initial proofs either by the end of this week or early next week. I ordered 5 of each thickness, 30, 20, 15 and 10 mil thick cards. Can't wait for them to come in! :D

qwertyaardvark, 2018-07-06:
So good news, the cards have arrived! YAY! Bad news... they are extremely loose when placed in the wheel. So much so, that the card can fall out on their own. I will fathom a guess that the badge puncher you use leaves a large radius at the bottom and that keeps your cards from falling out. My cards have very small radii and lets the cards move sideways * just * enough to fall out. I will look into having the die modified (or ouright re done) and adding back from material on the inside.

For reference, the total amount of gap that I am measuring is about 3.8mm total for both sides, so adding a 1.25 - 1.5mm will be more than enough to hold the cards, but also keep the cards better centered. Instead of the 4mm depth of the notch like what was drawn in #8, I will reduce the depth to 2.5 - 2.75mm. It's a shame it didnt work the first try, but I was thoroughly warned that this was an experimental build! I leave it up to you if want change the design to fit these cards, but I may have to change the design I use to avoid paying die costs again.

I can still send you what I do have so you can at least see the card/print quality. Just shoot me an email and let me know where to send it. :)

[image]
[image]
[image]

scottbez1, 2018-07-13:
Thanks for the update, and sorry to hear they didn't quite work out (but otherwise those flaps looks very exciting!). I'm thinking the dimension issue may have come from a combination of 2 sources of confusion:

First, I'm not sure where the 4mm notch depth came from in #8 - the badge punch I recommend punches a slot that's 15x3mm (and this matches the specification in the badge slot punch jig here https://github.com/scottbez1/splitflap/blob/6efafe5f2ae6161376d62a6f923a4bd91039beec/3d/punch_jig.scad#L28 which I've confirmed is accurate).

I think the second source of confusion is that the notch depth isn't explicitly specified in the 3d model; it's more or less assumed based on the thickness of the laser cut MDF (3.2mm) which just happens to work out conveniently with the badge slot punch 3mm slot width being close enough. I would consider this a bug though, and I'll try to update the .scad file soon to avoid using the MDF thickness as an assumed flap notch dimension. You can see the `thickness` "misuse" most apparently in the code for rendering the flaps here, where squares that measure `thickness` by `flap_notch` are cut out of the flap shape: https://github.com/scottbez1/splitflap/blob/6efafe5f2ae6161376d62a6f923a4bd91039beec/3d/splitflap.scad#L387

One question I had is that if the design assumed a 3.2mm flap notch, and my punch is only 3mm deep, wouldn't that mean my flaps shouldn't actually fit? However, the design has always had 0.5mm of "slop" built into it for the flaps to move side-to-side slightly within the spool (https://github.com/scottbez1/splitflap/blob/6efafe5f2ae6161376d62a6f923a4bd91039beec/3d/splitflap.scad#L105). This 0.5mm of slop is more than the (3.2-3)*2=0.4mm needed to make up for 3.2 vs 3mm discrepancy on each side, so in reality my prototypes have had ~0.1mm of slop rather than the intended 0.5mm.

So I'll try to clarify all of this in the .scad files in the next few days so that it's all explicit rather than coincidentally working out. My rough plan for now is to explicitly specify the flap notch depth as 3mm (not 3.2) and then update the `flap_width_slop` to be 0.1mm to compensate and match reality. When the dependent dimension definitions are updated as well, I believe the overall final dimensions of everything (like the spool and enclosure) should remain exactly the same, it's just that `flap_width_slop` won't be misleading any longer and the flaps should be rendered with the correct 3mm flap notch.

Thanks for bringing this stuff to my attention!

qwertyaardvark, 2018-08-02:
The new set of cards came in and they work beautifully! :D

https://photos.app.goo.gl/ioV8BsQTPgqFCV7U8

scottbez1, 2018-08-24:
By the way I've just placed an order for 5000 die-cut PVC flaps, and plan to make them available for purchase once they arrive. The samples turned out really nice:
[image]

vettorazi, 2018-09-02:
hey guys! where i can find the pdf with the card updated and ready for cut?
i download the first pdf of the post (https://github.com/scottbez1/splitflap/files/1976770/Github.Splitflap.Characters.pdf) but seems the wrong one, right?
thanks!!

mflmartin, 2018-09-05:
@scottbez1 Hi!, 

Where did you place the order for those split flaps? What thickness did you order? Is it PVC also?

Thanks!

mflmartin, 2018-09-06:
@qwertyaardvark, can you provide the supplier that prints and cuts those flaps like that? I want to see what are the options for ordering some.

Thanks!

scottbez1, 2018-09-08:
@mflmartin, I ended up finding a supplier on Alibaba after getting quotes from a number of US companies and companies abroad. The ones I ordered are PVC, 30mil (0.76mm) thick. I'm happy to share more details over email if you're just planning to order these for yourself, as I've so far been happy with the supplier I'm using (though I've only received samples so far. My first large order is in transit and won't arrive until next week).

I'm assuming you're interested in quite a few flaps if you plan to order them directly? The upfront mold fees were in the US $100-200 range from most suppliers I got quotes from, and per unit prices definitely get much cheaper with higher quantities in the multiple thousands. If you're in the US and just interested in a dozen or so modules' worth of flaps, I should have some available for sale online in the next week or so.

mflmartin, 2018-09-08:
Hi!

Wonderful, could you email me more details at martin [at] todologic.com ?
I have also a couple of other things that wanted to share.

Thanks so much!

scottbez1, 2018-09-18:
Quick update: the custom die-cut flaps I had made are now available [on Tindie](https://www.tindie.com/products/scottbez1/blank-split-flap-display-flaps/) if anyone is interested in professionally cut flaps. $50 for 4 modules' worth of flaps (165 flaps).

DouglasPearless, 2018-09-21:
Where did you get your custom die made, I am looking to have one made so I can cut a standard 6x4” photo so it is landscape rather than the portrait mode that your die uses ( if that makes sense? )

Cheers
Douglas

salcin23, 2018-10-14:
Great project! Thanks for all your work!

Is it possible to make the front from 3mm black acryl? Any design change necessary?

Did you think of ordering already printed flaps?

kevwhite, 2018-10-18:
@scottbez1 I've emailed you to request supplier details. It's for personal use

scottbez1, 2018-10-30:
Responding to a few past comments:

---

> Where did you get your custom die made, I am looking to have one made so I can cut a standard 6x4” photo so it is landscape rather than the portrait mode that your die uses ( if that makes sense? ) Cheers Douglas

@DouglasPearless I didn't actually get the die itself; I ordered the flaps from a plastic supplier who made a die for their own use, and they just ship me the final flaps, so unfortunately I'm not sure I can help you here.

---

>Is it possible to make the front from 3mm black acryl? Any design change necessary?
>Did you think of ordering already printed flaps?

@salcin23 I think you'd be mostly fine using 3mm acrylic rather than the 3.2mm MDF that's recommended without needing any design changes, but I can't say I've tried it myself. Two things I would note if you use acrylic for the spool and not just the enclosure:
- you may need to glue the spool pieces together; right now I've just relied on the press/friction fit of the MDF, but that may not work so well with acrylic that's slightly smaller than the 3.2mm specified in the design
- in the past when using opaque acrylic I ran into an issue where it transmits IR light rather than blocking/reflecting it as you'd expect. Not sure if that's true of most acrylics, but if so it would definitely be better to go with the newer "experimental" hall-effect sensor design rather (on the master branch) rather than the IR reflectance sensor design that relies on IR light only passing through a single hole in the spool (all current "stable" versions up to and including v0.4)

The other option is to update the `thickness` parameter in the CAD model and re-export things, but IIRC there's actually an assertion that fires if you lower it to 3mm due to some interactions between the thickness and flap sizes and tolerances, so that may end up being more work to correct that issue.

---

@kevwhite We've been in touch via email :)

---

Seeing as this thread has ended up covering multiple issues, and the initial discussions were more informational rather than actionable, I'm going to close this Issue for now as I don't think there's any remaining "task" to be done or question to be answered. Obviously I'm happy to continue discussing details of mass-producing flaps, but I think it will be helpful to have separate Issues for specific questions going forward.

closed 2018-10-30 by scottbez1

DouglasPearless, 2018-10-30:
@scottbez1 Thanks for the update on the die, I will design a small CNC blade based cutter to achieve the same thing :-)

whitebird, 2019-02-18:
@scottbez1 Where did you order your die-cut PVC flaps? I'm planning on building a 12 module counter so I'll need few.

As an alternative, I have access to a laser cutter where I generally use MDF and acrylic. I'm searching for a material that's similar in feel to PVC, but which I can cut.