scottbez1/splitflap #26: Peak Amperage & Overall Power Consumption

qwertyaardvark, 2018-05-20:
While receiving the parts for the project, I came across the power supply and I was wondering how many modules the 12V-6A power supply called out in the readme could safely support. Has someone measured the peak amount of amperage and total amount of energy the modules use when a module moves 39 characters forward (worst case regular operation, for example from 'B' to 'A')?

I am still waiting for my PCBs to arrive from seeedstudio, and would be interested in seeing the energy requirements of an individual module so that I could later extrapolate power requirements for large boards or determine the feasibility of a indoor/outdoor solar powered split flap display.

scottbez1, 2018-05-22:
I haven't done any scientific testing, but from some quick trials my power supply reads out ~0.7A max when driving 3 characters, so about 230mA per module? I would guess this is a bit averaged and the instantaneous current draw may be higher but I don't have a great way to test that at the moment. Based on the coil resistance and voltage (https://github.com/scottbez1/splitflap/wiki/Motor-info), it suggests a current of 338mA per module (12V/71Ω*2 (since two phases are powered at each step)).

So even if we assume 400mA per module, 6A should be more than enough to drive the max of 12 modules per Arduino (using 3 chained controller boards).

qwertyaardvark, 2018-05-22:
Thanks for the response! Even though not super scientific, these values will go a long way in sizing power requirements for different applications I had in mind. Again, much appreciated!

scottbez1, 2018-05-29:
I just looked at the [power supply part you specified](https://www.digikey.com/product-detail/en/cui-inc/SWI5-12-N-P5/102-4122-ND/6579912) again and it looks like it's actually 6W rather than 6A; it's rated for 500mA output at 12V, which would only be able to power 1 or maybe 2 modules.

I've been using a bench power supply so far, so I haven't looked into sourcing a cheap 12V supply quite yet, but something like https://www.amazon.com/LEDMO-Switching-Converter-Adapter-Transformer/dp/B01E6RMASC seems like it may be a good option for powering all 12 modules with 3 chained controller boards (note that the 12V supply is connected through the ribbon cable between controller boards, so only 1 power input to the first controller is necessary for all 3 chained boards to work).

qwertyaardvark, 2018-06-04:
Hah... mundane little details... but noted. 2 modules is all I can build in the near future, so my current wall supply will suffice for now. Will keep an eye out for a good 12V power supply.

scottbez1, 2018-09-04:
Closing for now since I think the question was answered

closed 2018-09-04 by scottbez1

whitebird, 2019-02-04:
According to the datasheet, the 2.1mm DC barrel jack (PJ-202A) is rated for 2.5A max: https://www.cui.com/product/resource/pj-202a.pdf

The allows chaining with a ribbon cable also with a current rating of 2.5A: https://www.digikey.com/product-detail/en/cui-inc/PJ-202A/CP-202A-ND/252007

But the PCB allows chaining of up to three controller PCBs connected to one UNO, totaling 12 modules. So 400mA * 12 = 4.8A which is above the rating.

This means that with one PCB you can drive about 6 modules when staying inside thee max 2.5A rating.

How would you make a bigger display with 9 modules?

whitebird, 2019-02-04:
Should the grounds be connected then?

scottbez1, 2019-03-23:
Looked into this a bit more today. [839-1514-ND](https://www.digikey.com/product-detail/en/tensility-international-corp/54-00131/839-1514-ND/9685440) looks to be a drop-in replacement for the barrel jack, with a higher current rating of 6A. I've compared the datasheet to the existing kicad footprint and all the dimensions I checked seem to match, so that's probably the easiest way to make the design work.

whitebird, 2019-03-23:
I think we need to look at the ribbon cable as well, I don't have the datasheet here, but I think it was also limited at 2.5A.

scottbez1, 2019-03-23:
Only 8 motors will be pulling current through the ribbon cable, so the max current in the first cable/connectors would be ~3.2A and ~1.6A in the second assuming 0.4A per motor max.

Both connectors are rated for 3A. Motor power and ground each have 2 pins assigned on the connector/cable, so the maximum current draw per pin in the fist cable should be ~1.6A.

The ribbon cable itself is 28 gauge, so the two main concerns there are voltage drop and heat generation. The cable specs 237 ohms per km max, meaning we can expect about .119 ohms resistance for the ~0.5m ribbon cable, or .237 ohms for the full circuit including the ground path. So there would be a voltage drop of ~0.76V on the first ribbon cable, and another ~0.38V in the second. This is potentially problematic if this reduces the torque too much. As far as heat goes, this is a much trickier topic, and different references provide different values, but considering split-flap usage is expected to be a fairly low duty-cycle operation, I don't expect it to be a problem (Table 4 of http://www.mosaic-industries.com/embedded-systems/electronic-instrument-design-new-product-development/cables/ribbon-cable-current-rating shows actual test measurements to be 2.6A capacity to maintain a 30C temperature rise on a similar cable. Table 1 suggests a lower 1.4A limit for chassis wiring. But both of these numbers refer to 100% duty cycle applications, and I suspect you're going to have other problems before ribbon cable heat if all motors of your split flap are continuously running).

Of course, I'm not an EE, so as with any hobbyist/open-source project, I'd recommend you come to your own conclusions on the safety and limits of this design.