scottbez1/splitflap #27: Gearless2 - Module-to-Module Connection

qwertyaardvark, 2018-05-20:
I was looking at the design for gearless2, and I noticed the holes that are directly above and below the axle holes for the v0.3 were no longer there. I presume the holes were for connecting modules together, but the gearless2 design lacks these holes? I can see that the lower holes had to go b/c the PCB was moved down there, but the upper holes likely could stay. Maybe the lower holes could be between the backstop bolt slot and the bottom plate holes?

Not sure if you had another idea in mind for connecting the modules, but I have added these holes back in my svg files.

scottbez1, 2018-05-22:
You're right - the connector holes were removed in 0ec4b1e - the issue is that with the updated design there's no longer room for a bolt head between flaps and the wall opposite the motor, so you couldn't actually put a connecting bolt there any more:
[image]

The other, smaller issue is that the walls were moved closer towards the flaps, so the distance between adjacent walls has increased to over 16mm, so either connecting bolts would require a separate line item on the BOM for a longer bolt, or else the "standard" M4 bolt would need to be changed to something quite a bit longer.

So for now I've omitted them while thinking through some alternatives, as I don't think it's a show-stopper to lose the connections for now (in an actual installation with multiple units there probably needs to be a way to attach the modules to the structure as well, not just to each other, so I'd want to tackle both of those issues at the same time)

One idea would be to laser-cut a "bridge" that could press-fit onto the top and bottom via some slots cut into the side walls, something like this:
[image]

Or it might make sense to somehow extend the top and bottom enclosure pieces horizontally so they meet the adjacent unit and then somehow allow those to be connected together. In any case, I haven't given this a ton of thought yet, so I'm open to other ideas.

qwertyaardvark, 2018-05-22:
On a smaller scale and based on the assumption people will make multiple modules, I am planning on making the front bezel a single piece for multiple modules. I can fit up to 9 module bezels in a row in a Ponoko P3 sheet, but will likely reduce it to 8, as that's how many modules I can fit on a P3 sheet anyways. Alternately, I may make 2 4-character bezels to match the capacity of a controller. I would leave the nodes for individual modules so that smaller or individual bezels can be made.

Past 8 modules, there still needs to be way to connect between modules. I personally think there is still enough space between the flap exclusion zone and the top enclosure piece for a longer connecting bolt. It would require a new part on the BOM, so I will think about other mounting methods. For me, it wouldn't be a big deal to source a longer bolt, and that is what I'll go with for now. I do like your bridge idea, as its much easier to apply and remove. 

For one of my applications, I intend on not having a structure to save material costs and will mount directly to the modules at their bases (1 row x 12 modules). For the application of a large board (4+ rows x 20/24 modules) that will hang off a wall and needs a structure, I am torn between these two choices:
* integrate the overall structure into each module
  * higher material cost per module
  * more difficult to replace a single module
  * simpler to design/assemble module-to-module connection
  * highly modular number rows/columns
* making a separate structure that will hold the modules 
  * has to be custom designed for a size
  * complexity of structure and module-to-structure interface design
  * overall lower material cost and weight
  * easier to replace a module

I would like to go the route of making the structure integrated into each module. Will keep the concept in my head and let you know what I come up with.

scottbez1, 2018-06-04:
My plan right now is to add two related connection components to the design, which are designed to allow the modules to mount to an external structure with proper alignment and spacing. It will _not_ enable modules to be directly connected to one another to form a rigid structure on their own.

The reason for this is that an external structure can be as simple/cheap as a rectangular plywood base board for them to be screwed down into. It also means it will be much easier to access/service individual modules.

My planned changes are:
- add a simplified version of the "bridge" connector I mentioned above. I plan to have one at the top and one at the bottom. This will press fit (or may even be loose and simply sit in place depending on tolerances of laser cutting, but that's fine) and is only intended to maintain spacing and alignment in the XY plane
- add a single 4mm/M4 hole, centered in the X dimension, in the top and bottom walls, which will allow the modules to be mounted to a board above or below if desired. Having just a single hole would allow rotation about the Z axis on its own, but the bridge connectors should prevent that rotation

On a larger display, I'd imagine the best way to mount modules is probably to have a single front panel for a group of modules (e.g. something like 2 rows x 8 columns per front panel, depending on how many can fit in a single laser-cut), and each module mounts to that front face simply using the existing 2 front screws + tabs and nothing else. That would also allow individual modules to be serviced or replaced by removing the 2 screws and pulling it out from the back. The removed module won't have the rigidity provided by the front face, so could potentially parallelogram a bit, but that's probably not a huge deal; you could always slip on an individual front face to hold everything square while testing/repairing the innards of a single module.

scottbez1, 2018-06-04:
I've pushed the initial changes to the 3d design in ad89cf78a07d8c13f3687d7be0f420d4e2130e40

The connector bracket fits in place like so:
[image]

It's held in place by 2 locking tabs that stick inward and latch into a small hole on the inside of the wall. Here's what the bracket looks like when it's not in place - you can see the locking tabs a bit better:
[image]

scottbez1, 2018-06-06:
I landed the 2d update in b20415945d8a6b9d5907e3c98b3b7a8898dec790 and am awaiting a new order from Ponoko. Closing this for now but will reopen if any issues come up when I get the new design in a few days.

closed 2018-06-06 by scottbez1