scottbez1/splitflap #28: 0.4v? no updated release

killboxalpha, 2018-07-13:
I see the combined preview and splitflap.scad have been updated but no new release. as someone who hates openscad with a passion and wants to use fusion i need a svg or dxf to modify files, I am working on a smd version of the board and want to use the gear-less version of this, but the files seem to be missing? 

I'm trying to export dxf/svg's from scad but it seems to be a very different scale.

killboxalpha, 2018-07-13:
forgot to read the read-me

killboxalpha, 2018-07-13:
I can not for the life of me get the .py script to work, it keeps complaining about a missing module for svg.path?

scottbez1, 2018-07-15:
You'll need to install the svg.path module (e.g. `pip install svg.path`) before running the script. Though I should also note that all of the instructions in the readme are intended for linux, so if you're on another operating system they likely still won't work for other reasons.

I'll try and upload the generated files to the 0.4 release in the next day or so, thanks for bringing this up!