scottbez1/splitflap #32: Provide example Arduino code for basic single-module breadboard configuration (no shift register)

scottbez1, 2018-07-31:
For ease of experimentation, especially for hobbyists who don't have the experience or desire to order custom PCBs, it would be nice to provide some sample Arduino code for driving a single splitflap module using a basic standard ULN2003A control board rather than the shift-register based custom PCB.

It should be pretty easy, as the SplitflapModule just operates on half of a uint8_t buffer for motor output and a single bit of a uint8_t buffer for sensor input:

```
  SplitflapModule(
    uint8_t &motor_out,
    const uint8_t motor_bitshift,
    uint8_t &sensor_in,
    const uint8_t sensor_bitmask
  );
```

```
inline void SplitflapModule::SetMotor(uint8_t out) {
  motor_out = (motor_out & ~(0x0F << motor_bitshift)) | ((out & 0x0F) << motor_bitshift);
}
```

```
bool SplitflapModule::CheckSensor() {
    bool cur_home = (sensor_in & sensor_bitmask) != 0;
    bool shift = cur_home == true && last_home == false;
    last_home = cur_home;

    return shift;
}
```

It may even be possible to just instantiate a `SplitflapModule` instance by directly passing e.g. `PORTD` as the motor_out reference and e.g. `PORTC` as the sensor_in reference, since SplitflapModule already safely writes all 4 bits to the motor_out reference in a single operation.

Would also be nice to provide a Fritzing circuit/breadboard diagram on the wiki for how to wire it up.

daniele8808, 2018-08-17:
Hi Scott,
I'm from Italy and I'm a maker and interaction designer.
I am really enthusiast for this project ( I always liked this kind of mechanical projects) and I would like to make my personal version of it.

I made two split flap units using your wiki documentations and one friends inside a Fablab (to make the laser cut  enclosure) and I've also got all the electronics( steppers, controller boards, Arduino, jumpers and 2 reflective sensors)

I would like at the beginning try to control just one or two modules without developing your custom PCB but simply using one Arduino and some jumpers.

It would be very nice and helpful for me to see how to make the connection between the Arduino's pins and the motors controller boards and the sensors.

Could you please share some more information about this connections without using your custom PCB?

Any help would be really appreciated.

Thanks in advance, and congratulations for this project. 
It is awesome.

Daniele

scottbez1, 2018-08-20:
Hi Daniele,

Thanks for the feedback and motivation to address this issue! It's always fun to hear from people around the world.

I've just pushed some updated Arduino code that should make it easy to use the "standard" ULN2003A-based driver boards that frequently come with the stepper motors in lieu of a custom PCB. I'm planning to use Fritzing to create a wiring diagram soon, but in the meantime the basic pin connections are roughly documented here: https://github.com/scottbez1/splitflap/blob/4a797a385efdf51649ecae620c819b7a86b21bc7/arduino/splitflap/basic_io_config.h#L32 To use that new basic IO mode, you'd set `NUM_MODULES` to 3 or fewer, `NEOPIXEL_DEBUGGING_ENABLED` to `false`, and `SPI_IO` to `false`.

On the other hand, if you already have the MIC5842 and SN74HC165 chips as needed for the custom PCB, you could also wire those up on a breadboard based on the [schematic](https://s3.amazonaws.com/splitflap-travis/branches/master/schematic.pdf), though there will be more wiring required for this approach.

I hope to have the basic wiring diagram done later tonight or tomorrow.

scottbez1, 2018-08-21:
The wiring diagram and description has been added to the wiki: https://github.com/scottbez1/splitflap/wiki/Electronics#basic-prototyping-alternative-electronics-approach

closed 2018-08-21 by scottbez1

daniele8808, 2018-08-21:
Hi Scott, 
thanks for your reply. 

As soon as I come back to my place I will try it and I can't wait to see it working.
Unfortunately at the moment I have just an Arduino Yun which has a different processor and it seems not working.

I will update when I will test it with my other Arduino.
Thanks in advance for your help.

vettorazi, 2018-11-12:
Hey Scott and Daniele, i am following this issue and i just tested with a arduino micro.
First i used a MEGA, and it was the same issue (different processor)
Then, with a micro board, worked just fine.
question:
Can i use more than 3 modules with an arduino mega?
and has some way to use more modules with more arduinos? or just ordering/making the pcb?
can i connect more arduinos (via i2c, idk) to accomplish more modules?