scottbez1/splitflap #33: More Instructions about the IR sensor

daniele8808, 2018-08-26:
Hi Scott I've been reading all the wiki and also the instructions on the imgur websites but I still have some problems about the IR sensor. 

I have two questions about it:
Do I have to add some "sign of recognition" or some "marks" on the gear to make the sensor understand the home position? 
Here some picture of my prototype:
[https://ibb.co/hJeLTU](url)
[https://ibb.co/gifG2p](url)
[https://ibb.co/fqMOhp](url)

I saw on IMGUR websites that you make a second hole on it but I don't see it on the Illustrator file for lasercutting machine.
Do we need to insert something in the hole to make it recognizable by the sensor?

I make a hole using a screwdriver but it seems not recognized by the sensor so I added a screw inside it and the sensor now seems work a little bit better but not as expected.

The second question is about the code. I place this screw close to the "e" letter but this is not what I wanted,  I would like to set the home position to the "a" letter.
Is there a way to set the home position by code or do I have to replace all the cards to make the home position to the letter I desire?

scottbez1, 2018-08-29:
>Is there a way to set the home position by code or do I have to replace all the cards to make the home position to the letter I desire?

While you _can_ update the code to adjust which letter is in the home position, I would generally recommend moving the cards so that the letter you want is in the home position. Though of course I would wait to do this until the IR sensor issue is resolved because you wouldn't want to have to do it twice 😛

The reason for that is that the Arduino code assumes all modules have the same home position, so it's _VERY_ convenient if all modules are physically configured so the letter you want is aligned with the sensor's home position. It would be possible to configure a separate offset for each module in the code, but then you'd always have to make sure you don't mix up which module is plugged into which ports or else the wrong offsets would be applied.

If you do want to adjust the home position (or really anything about how the flap order is defined in the code), you can do that by rearranging the items in the `flaps` array in the main sketch:

https://github.com/scottbez1/splitflap/blob/master/arduino/splitflap/splitflap.ino#L34

The "home" position is the first item in that array.

>Is there a way to test if the IR sensor is correctly working?

There are a few steps that I would take to debug this problem:
1. **Test the motor/mechanics in isolation:** I saw what looked like a few hesitations or possible missed steps in your video, so figured I'd suggest this as a starting point. I'd first disable the home position sensors entirely by setting `HOME_CALIBRATION_ENABLED` to `false` in `splitflap_module.h` -- the reason for this is that a flaky/broken sensor can cause all sorts of strange behavior. Then, use the Serial Monitor to tell the Arduino to go to the empty space (i.e. `= \n`) - check to make sure that the flaps turn smoothly, and remember what letter the module ends on. Do the same thing again and confirm that the module ends up on the same letter. Do this a few more times (at least 5) to make sure everything is working mechanically. If it doesn't return to the same letter and you've verified that there's nothing physically obstructing the spool from turning (e.g. sometimes flaps will get caught on the top/sides of the window), let me know and I can provide some additional troubleshooting steps.
2. **Test the sensor in isolation:** If you have a volt meter (or ideally an oscilloscope), connect it to the sensor's output pin while the sensor is plugged into the Arduino. The voltage should be near 0V (or at least lower than 3V) when the sensor is NOT aligned with the hole. Then manually rotate the flaps carefully until the hole is aligned with the IR sensor (on the design with gears I'd recommend rotating the motor gear rather than the spool itself). The voltage should increase to be near 5V. The change in voltage should be very obvious; if it's not, it suggests there's not enough contrast between the home position and everything else. If that's the case I would suggest cutting some white paper into a circle, making a black dot or line with a permanent marker, and then taping it to the gear's surface. If you don't have a voltmeter or oscilloscope, you can connect the sensor to an analog pin on the Arduino and check the voltage using a simple sketch that outputs the analog value over serial, though _you'll need to make sure to enable the internal pull-up if you do this_ (i.e. `pinMode(X, INPUT_PULLUP)`).

Hopefully one of those two steps is useful in tracking down the issue.

By the way, I noticed in your video the "count_missed_home" value increases when the display seems to have trouble, while the "count_unexpected_home" value remains at 0. When "count_missed_home" increases, that means the controller expected to find the home position (once per full revolution) but rotated past where it was expected and still didn't find it. When that occurs, the controller immediately enters calibration mode, where it rotates slowly to find the home position, and then once the home position is found it will rotate at full speed to the last requested letter. This generally happens if the motor got jammed up at some point and didn't rotate the correct amount, or else the sensor just didn't find the home position for some reason (e.g. it's not aligned with the hole correctly) -- this is why I've suggested the two troubleshooting steps above.

scottbez1, 2018-09-04:
Hey @daniele8808, any luck tracking down the issue? Happy to continue helping if you're still running into issues, otherwise I'd like to go ahead and close this issue (I've added the troubleshooting steps from above to [the wiki](https://github.com/scottbez1/splitflap/wiki/Troubleshooting) since I'm certain you're not the only one who has run into issues like this).

daniele8808, 2018-09-19:
Hi Scott I've been out for work until today.
I will try what you wrote this afternoon and update you soon. 
Thank you.

scottbez1, 2018-10-23:
Closing this for now. Feel free to reopen if you're still having issues.

closed 2018-10-23 by scottbez1