scottbez1/splitflap #37: Split Flap Card, cut and stickers

l0l0l0, 2018-09-18:
Hello, 
Congratulations for your wonderful project !
I was wondering if it would not be easier to put the sticker before cutting the cards?

thanks

scottbez1, 2018-09-19:
That could theoretically be done, but would only be possible for every second letter since flaps are double-sided, so I'm not sure it would help all that much?

By the way, I recently updated the wiki with a new recommended way of applying the letter stickers, which should be a lot quicker and more precise than the old method of cutting the stickers in half and applying one half at a time: https://github.com/scottbez1/splitflap/wiki/Apply-flap-stickers#new-instructions

scottbez1, 2018-09-20:
Closing for now. Let me know if I've misunderstood or missed something.

closed 2018-09-20 by scottbez1