scottbez1/splitflap #38: Documentation on ordering parts needs improvement

scottbez1, 2018-09-21:
As @mflmartin helpfully highlighted to me, the documentation around ordering the electronics components is somewhat outdated, and has a number of assumptions baked in that may not be obvious.

I see two main issues:
- The relationship between the number of modules, the number of controller boards, the number of sensor boards, and the amount of supporting hardware needed (Arduinos, power supply amperage, etc) is unclear. This needs its own wiki page, ideally with supporting diagrams of different example cases (HELP WANTED for good looking vector graphics for this!)
- the electronics BOM is auto-generated, but doesn't clearly explain what components it includes and whether all of them are actually necessary. I.e. it currently includes all components that might be needed for either an A, B, or C variant controller board (when in reality you only need some subset, depending on which variant you're making) and enough parts for 4 sensor boards (which you may not need if you're making fewer than 4 modules). It may make sense to simplify things to 2 variants: a "shield" ("A") and "expansion" (currently "B") variant. It may also make sense to either auto-generate an Excel spreadsheet with formulas for calculating part quantities based on # of modules, # of "shield" variant boards, and # of "expansion" variant boards, or else maybe create a BOM wizard on the project website to do those calculations based on the generated .csv?

martinlombana, 2019-01-20:
I also wanted to know if we must modify something for 3mm sheets, as I want to do them in acrylic. 

Thanks!

scottbez1, 2019-01-22:
Oh, interesting! Let me know how it turns out! I may try ordering some 3mm acrylic as well just to see how that works

scottbez1, 2019-01-31:
@whitebird - are you still seeing that permission error? I just tested the link in an incognito window and it seemed to work fine. 

If you're still hitting the 403 error - are you logged into a Dropbox account? Can you try opening it in an incognito window if so?

whitebird, 2019-02-01:
@scottbez1 It's indeed only an issue when I'm logged in. Thank you!

ArwedSchmidt, 2019-02-15:
@scottbez1 The MDF parts arrived (some delay due to customs over here in Germany) and I gave it a first try - looks like 3.0mm is just fine. Everything was a kinda tight fit so I had to use some gentle force but I guess that's how it's supposed to be

ericyeargan, 2019-07-16:
Hmm, 3.2mm MDF no longer seems to be available in the US either: https://make.ponoko.com/materials/mdf-fiberboard

Oddly enough the ["show-material" URL](https://www.ponoko.com/make-and-sell/show-material/64-mdf-natural) that @scottbez1 posted _does_ show 3.2mm but I suspect that page is no longer valid - it redirects to the main page if you're logged in.

@ArwedSchmidt, when you ordered 3.0mm MDF, did you modify the SCAD to use 3.0mm thickness and regenerate the SVG or just submit the 3.2mm release version?

scottbez1, 2020-04-18:
Closing this issue for now. Keeping #76 open to track the 3.0mm MDF documentation update.

closed 2020-04-18 by scottbez1