scottbez1/splitflap #42: svg processor doenst work?

vettorazi, 2018-10-23:
hey guys, i am trying generate the svg from the py, but it shows:
splitflap znb$ 3d/generate_2d.py
Traceback (most recent call last):
  File "3d/generate_2d.py", line 27, in <module>
    from svg_processor import SvgProcessor
  File "/Users//splitflap/3d/svg_processor.py", line 16, in <module>
    from svg.path import (
ImportError: No module named svg.path
:splitflap znb$ 3d/generate_2d.py
Traceback (most recent call last):
  File "3d/generate_2d.py", line 27, in <module>
    from svg_processor import SvgProcessor
  File "/Users//splitflap/3d/svg_processor.py", line 16, in <module>
    from svg.path import (
ImportError: No module named svg.path

scottbez1, 2018-10-24:
Hey there. It looks like you need to install the svg.path python library. Assuming you're on Ubuntu (which is the only supported OS at the moment), you'll want to run something like `sudo pip install -r "3d/scripts/requirements.txt"` to install the necessary python dependencies.

scottbez1, 2018-10-29:
Did that work? Closing for now, but please reopen if you're still running into issues.

closed 2018-10-29 by scottbez1

whimsee, 2022-12-26:
Somewhat related since I also ran across this issue from a different angle (MutableMapping didn't import which lead to this error) and even went and VMed the latest Ubuntu to see if it's OS-related, but that didn't fix it. Turns out it's apparently related to [https://stackoverflow.com/questions/70870041/cannot-import-name-mutablemapping-from-collections](url) because direct imports has stopped working since Python 3.9.

So I edited path.py ("from collections import MutableMapping" to "from collections.abc import MutableMapping") and that did the trick