scottbez1/splitflap #43: multiple versions on the download area. WHY?!?!

vettorazi, 2018-10-23:
Hey, why are two main versions of the mechanics, in the download area?
one, in the zip, and other in a single svg?
it's very frustating, after spend some days cutting the mechanics, discover that is a slight different version in the same release.
https://github.com/scottbez1/splitflap/releases/download/v0.2-mechanics/combined.svg
https://github.com/scottbez1/splitflap/archive/v0.2-mechanics.zip

scottbez1, 2018-10-24:
Can you clarify what you're seeing as different?

The zip is the source code of a release, not the rendered artifacts, so this generally wouldn't be used for anything (GitHub creates this zip and automatically includes it in releases, so there's not too much I can do there).

The other assets, like combined.svg are the proper artifacts that should be used for a given release.

scottbez1, 2018-10-29:
Did that response make sense? Closing this issue for now, but please let me know or repoen the issue if you still have questions about the releases.

closed 2018-10-29 by scottbez1