scottbez1/splitflap #48: v0.5 Laser parts offset

henricazottes, 2018-11-12:
Hi,

I cut a set today and after assembling parts, I see a small offset of the wheel regarding the front hole:

[image]
[image]

The blue space blue seems fine but the red one seems a bit too much. Flipping the front panel was even worth. Is the assembly incorrect or is there an error in the file ?

Thanks

scottbez1, 2018-11-12:
It looks like you have the spool outer disk and the inner square piece
(that the bolt attaches to) swapped - the smaller square piece should be on
the inside of the spool disk.

See https://youtu.be/zHM8W6Rm2i4?t=136 for assembly instructions (sorry I
haven't yet linked to this video from the project wiki).

henricazottes, 2018-11-13:
Thanks, I could have guessed it...

Another question: the video made me realize that magnets are missing from parts list. What ref should I look for ? The hole is 3.7mm wide but it didn't give me a lot results.

scottbez1, 2018-11-14:
It's a little odd, but the magnets are actually in the electronics BOM since I originally used this magnet from Digikey -- [469-1027-ND](https://www.digikey.com/product-detail/en/radial-magnet-inc/9027/469-1027-ND/5218822) a 4mm diameter x 2.5mm neodymium magnet.

I've also had success with some cheap 4mm x 1mm round neodymium magnets from AliExpress as an alternative. You could probably also get away with a small cube magnet (gluing it in place if needed) or drilling the hole bigger and using a bigger magnet. As long as it's strong enough for the hall-effect sensor it shouldn't matter too much. And the hall-effect sensor can be moved within its mounting slot, so there's nothing particularly precise about where the magnet is mounted on the spool; the sensor can be shifted forward or back as needed to fine-tune.

scottbez1, 2018-11-21:
Closing this issue to tidy up the issue tracker since I think I answered your questions, but please feel free to open a new issue if you have any more questions, more than happy to help :)

closed 2018-11-21 by scottbez1