scottbez1/splitflap #49: more arduinos instead of  expansion board

vettorazi, 2018-11-14:
Hey guys,
i already cut all the parts, mounted all together and I already pasted the vinyl.
I am thinking about the eletronics parts. and it's kind of expensive and takes too long to ship to brazil. 
so i am wondering... Can i use more arduinos instead of the expansion board? has some way?
if i use the MEGA, could be possible to put more modules?
thanks!

scottbez1, 2018-11-21:
Hmmm, yeah it should definitely be possible to use a mega to control more modules (from a quick glance I'd expect you could do at least 10), though I haven't tried it myself.

It will require a bit of customization of `basic_io_config.h` and some investigation of the Mega's pinout, but nothing terribly complicated. Are you familiar with raw port manipulation on Arduino? i.e. using things like DDRB, PORTB, or PINB rather than digitalRead/digitalWrite? If not I can walk you through the process of adjusting that file and figuring out which pins to use on a mega to make it work, just let me know!

vettorazi, 2018-11-22:
Hey scott! Thanks for your reply!
I am testing use I2C communication, using some cheap Pro Micros as slaves.
i might have some result later today.
I will post the results here.

About your question abou raw port. I am not familiar. But i will search
about. Good tip!

vettorazi, 2018-11-23:
Hey @scottbez1 
i've managed to get other arduino 'talk' to the arduino controlling the flaps, via Serial.
But i am having a trouble to find the answer to my problem:
(just for test) my arduino mega are sending the same command over and over again:

> '=hel'

(in the monitor serial, i see the bytes normally:
61 (=) 
98(a) 
99(b)
100(c)
10(enter)

the slave (arduino controlling the flap) are receiving the bytes normally.
But are not moving the flaps.

What could be? any tip?

scottbez1, 2018-12-03:
Just catching up on the issue tracker. Can you provide a little more info about how you've connected the Arduinos together if you're still running into issues - which pins are connected to each other?

scottbez1, 2020-04-18:
Closing old issue pending response.

closed 2020-04-18 by scottbez1