scottbez1/splitflap #50: Motor rotating the wrong way

henricazottes, 2018-11-21:


vettorazi, 2018-11-21:
hey! did you checked the pins from the board to the arduino? the motor are in the right side (forward to you?)

scottbez1, 2018-11-21:
I've gotten one batch of motors that turned the opposite direction for whatever reason. These things are so cheap and I think manufactured by a bunch of different companies that you tend to get slightly different products from different vendors. That said, you can change the direction in code fairly easily by reversing the elements of the `step_pattern` array in splitflap_module.h:

https://github.com/scottbez1/splitflap/blob/ec1b76c90de9fa56e2e0075ef8a7e6e88a0664d0/arduino/splitflap/splitflap_module.h#L164

Another thing to keep in mind is that there are two different gear ratios I've seen in motors, as mentioned on the [Motor Info](https://github.com/scottbez1/splitflap/wiki/Motor-info) wiki page, so you may need to change the `_GEAR_RATIO_INPUT` and `_GEAR_RATIO_OUTPUT` values as described in the comment here: https://github.com/scottbez1/splitflap/blob/ec1b76c90de9fa56e2e0075ef8a7e6e88a0664d0/arduino/splitflap/splitflap_module.h#L36

The non-integer 63.684:1 gear ratio seems to be much more common in the 5V version of motors, so you may want to check for that as well if the motor doesn't seem to be aligned right after several full revolutions.

And one question out of curiosity - is the order of wires on your motor connectors Red, Orange, Yellow, Pink, Blue?

henricazottes, 2018-11-23:
Thanks, reversing the `step_pattern` did the job.
``` c++
const uint8_t step_pattern[] = {
  MOT_PHASE_D | MOT_PHASE_A,
  MOT_PHASE_C | MOT_PHASE_D,
  MOT_PHASE_B | MOT_PHASE_C,
  MOT_PHASE_A | MOT_PHASE_B,
};
```
Didn't have to touch gear ratio. And yes, colors are in the order you mentioned :)

scottbez1, 2018-12-03:
:+1:  Going to change the title of this issue to make it a feature request for adding a direction toggle to the code. It really should be easier to flip the direction than reversing an array.

commit 55e353b5 (2019-01-05): Move sensor_test into splitflap sketch

This eliminates duplication of splitflap module IO code, which means you should now be able to run sensor tests in basic_io mode (previously sensor_test assumed spi_io mode). This is helpful for ESP8266 support (#55) so that latch pin configurations apply to sensor tests as well as regular run mode.

Also, this fixes #50 by adding a new `REVERSE_MOTOR_DIRECTION` flag.
closed 2019-01-05 by scottbez1