scottbez1/splitflap #51: Ethernet shield compatibility

henricazottes, 2018-11-23:
I'd like to have the screen connected to ethernet so I can control it from the network. From what I see, it uses the spi bus from the ICSP connector. This prevent my ethernet shield from being compatible because it also need those SPI lines (I' using a clone of this board: http://shieldlist.org/arduino/ethernet-v5). From what I read it could be possible to have it working by using SS pins (I have no experience with SPI bus). I'll to work it out but if you have any tips for it feel free to share ;)

henricazottes, 2018-11-29:
Ok so I'm definitely not comfortable enough with SPI and your project to choose which direction to take... But here are the two options I found:
- Using hardware SPI implementation with SS pin
- Using an external library to create a SoftSPI (like https://github.com/niteris/ArduinoSoftSpi or https://github.com/greiman/DigitalIO)

Do you know if any of these options could be possible ? And if yes, which one do you think is the easiest to implement ?

henricazottes, 2018-12-03:
Ok, that's what I thought. Thanks for the detailed explanation!
As a solution I ended up using a second Arduino with the shield and made the two of them communicate over UART. Working great, but it's not the best form factor :p

scottbez1, 2018-12-03:
Ah yeah, that's a great solution! Glad to hear it's working, even if it's not the prettiest thing :)

I'm going to leave this open but assign it low priority, as it seems like a valid concern on a somewhat rare combination (anecdotally I've heard a lot more interest in ESP8266-based wifi integration, so I'm leaning toward focusing on that as far as network integration is concerned).

scottbez1, 2020-04-18:
Closing this for now - no current plans to support Ethernet shields (though pull requests are always welcome!)

closed 2020-04-18 by scottbez1