scottbez1/splitflap #52: home sensor

carlosbritotx, 2018-11-24:
Hi friend, could you explain the code for your home sensor? I'm using a nema 17 engine instead of yours. I want help to understand the code of your home sensor. Could you explain me. Congratulations on the project.

scottbez1, 2018-12-03:
Here's a really rough summary of what all the SplitflapModule does to drive a splitflap motor and account for home sensor feedback:

It drives the motor based on a commanded flap index, which requires keeping track of the current step/phase as well as the motor position, and calls to update() check to see if it's time to advance to the next step (based on the distance to the desired flap and an acceleration curve from acceleration.h).

This gets somewhat complicated because the 28BYJ-48 motors sometimes have non-integer gear ratios resulting in a non-integral number of steps per spool revolution. Because of this, the motor position is tracked in terms of steps (`current_step`), which can be converted to a spool position by dividing by the gear ratio. Making matters worse, however, is the fact that floating point math is expensive on the Arduino, so the code has been optimized to use integer math instead (which generally means keeping a numerator and denominator as separate integers rather than calculating a floating point quotient). This is why there's quite a bit of additional confusing math throughout.

The way the home sensor interacts is that once per revolution the code calculates the step index where it expects the home sensor to be triggered (`expected_home`) and from that it calculates 3 important indices: `unexpected_home_start_step`, `unexpected_home_end_step`, and `missed_home_step`. The "unexpected_home" values are a range of step indices where it would be "unexpected" for the home sensor to be triggered (i.e. something weird happened and the spool got to the home position EARLIER than expected). The `missed_home_step` is calculated to be  slightly after the `expected_home` position; if the controller gets to that step without the home sensor being triggered then something went wrong and the spool didn't rotate far enough. For reasons that are slightly complicated (related to the gear ratio stuff mentioned above), all of the step indices are computed _modulo_ 2 * GEAR_RATIO_INPUT_STEPS, which you may see scattered throughout the code.

In any case, I think to support a NEMA 17 motor (assuming your motor controller has STEP/DIR inputs) you could probably modify `STEPS_PER_MOTOR_REVOLUTION` to be the motor's rated steps per revolution (which is probably something like 200) and then set `_GEAR_RATIO_INPUT` and `_GEAR_RATIO_OUTPUT` both to 1. Then you'd want to remove the SetMotor method and replace where you see `current_step++` with a pulse of the STEP pin to the motor driver. In theory all of the sensor code should still just work, but I've never tried a different motor like this.

carlosbritotx, 2018-12-03:
Thanks for the comment, I'm taking the first steps on the arduino. I'm using the a4988 and nema 17 4-wire drive, I'm kind of confused in the code, esou using two 20-tooth pulleys and a strap, my reel has 30 tabs, could you help me with an example code, is not able to use your code here on my. Many thanks friend.

closed 2020-04-18 by scottbez1