scottbez1/splitflap #55: ESP8266 support

scottbez1, 2018-12-13:
I've heard interest from a number of people about controlling a splitflap from wifi, and ESP8266 is the obvious default choice. It would be cool to see what's needed to get the code up and running on ESP8266 rather than an Uno, ideally with no hardware changes to the control board (at least initially; long term I could see a new revision of the control board that either takes an ESP board directly or plugs into something like a wemos d1 mini).

I spent a little time a while ago and got basic motor output working over hardware spi, but I recall having issues getting the sensor input working correctly over hardware spi but didn't investigate further. I also forget if I was running the control board at 3.3v or 5v.

In any case, opening this issue to centralize discussions around progress or investigations (cc @johnboiles)

NilSkilz, 2018-12-18:
I've got a version of the code working with an ESP8266 at the moment. It's a little awkward as the module's RX/TX goes to pin 2/3 of the Arduino.  It should be fairly simple to alter the PCB to accommodate it. 
I then just send TCP commands to it and your code does the rest.

scottbez1, 2018-12-21:
Played around with this a little bit today, and it's getting close, but not fully working.

Here's how I have the connections from the bottom of the control PCB hooked up to my Wemos D1 Mini:
[image]

A few initial code modifications needed:
- add a `yield()` to the beginning of the `while(1)` within `loop()`, otherwise Soft WDT fires after a few seconds (alternatively could remove the inner loop, though on an Uno I think this may add extra overhead to the loop if `loop()` doesn't get inlined and it's actually making a function call each iteration)
- change `Adafruit_NeoPixel` config to use pin `D3` rather than `6`
- change `spi_io_config.h`:
  - set `OUT_LATCH_PIN` to `D1`
  - set `IN_LATCH_PIN` to `D2`
  - modify `motor_sensor_io` to use `digitalWrite(IN_LATCH_PIN, ...)` etc rather than direct `PORT` manipulation (need to double-check this isn't too slow, but considering the ESP may be up to 10x faster than an Uno I'm not sure it matters)

Driving the motor seems to work fine, but reading the sensor input from the 74HC165 seems to be shifted by one bit. There seems to be some discussion around the ESP8266/Arduino core SPI implementation and clock polarity (e.g. https://www.reddit.com/r/AskElectronics/comments/6d7sby/problems_connecting_74hc165_piso_shift_register/ ) so the next step will be some experimentation/digging to figure out what's going on here. In the meantime, I can get a single module working fine if I plug the motor into port B and the sensor into port A on the control board (compensating for the sensor input being shifted by one).

whitebird, 2019-03-05:
Have you also looked at supporting its successor, the ESP32?

scottbez1, 2019-03-18:
@whitebird I'm interested in the ESP32 as well (mostly because it appears to have proper HTTPS cert validation rather than the fingerprint-based implementation on the ESP8266 which breaks whenever the leaf cert is renewed/rotated). 

Are there any solid, cheap development boards with USB built in, similar to the Wemos D1 Mini? I've got a few ESP32 dev boards, but so far I haven't seen any particular board take off the way the Wemos did for the ESP8266, and the confusion between two variants of the dev board with different pinouts marketed under the same "devkit v1" name makes me hesitant to design around that for now.

The other alternative I've thought about would be to just design a board you can solder the ESP32 module to directly, but then people would need a separate programmer to flash the initial image with. The design would also need to include the 3.3v regulation and other supporting components, which add to the cost (this is one of the best parts of the Wemos D1 Mini in my opinion; it's cheaper to buy the Wemos than it is to buy a bare module and the USB connector, voltage regulator, and other passives).

scottbez1, 2019-03-18:
The changes in c119432 fix the off-by-one-bit issue. Still saw a Soft WDT reset which is worrisome. I need to dig into the behavior when that happens and figure out how to make that safe (i.e. make sure to reset the motor drivers to avoid leaving coils energized indefinitely if the WDT fires during movement) and/or prevent it from happening altogether before I'm comfortable considering this ESP8266 support complete.

johnboiles, 2019-08-10:
I wonder if there are just more places we can add a `yield()` to prevent the wdt. I’m hoping to give this a try soon so I can finally put my splitflap up on the wall