scottbez1/splitflap #56: Add Project Gallery In Wiki

tknickman, 2018-12-19:
Now that a couple of kits have gone out, it could be cool to add another sidebar on the Wiki for users to add photos of their completed split flap projects (either from the kits, or from ordering their own parts)!

I think it would be pretty helpful for people embarking down this road to see how others have faired / see what variations people have added!

I will be working on mine over Christmas, so happy to kick this off after I have my finalized project if you think it's a good idea.

What do you think? (Love the project by the way :) )

scottbez1, 2018-12-19:
I think this is a great idea! Definitely feel free to get it started when you have some photos to share - the wiki should be editable by anyone.

The only thing I'm not totally sure about is image hosting. One of the reasons I've been trying Dropbox Paper for some of the newer documentation is that it's so much easier to embed content than the GitHub wiki. It's possible to clone the wiki as a git repo and add images that way, but it's somewhat of a pain to do that so I'm open to alternatives. Maybe a Paper doc could work for this too?

tknickman, 2018-12-20:
Awesome sounds good!

True, image hosting makes the exact details of this a bit ambiguous, I can add a section in the soon-to-be wiki about hosting options.

tknickman, 2018-12-20:
Closing for now as an approved idea to keep your issues clean, I'll ping you again when I'm closer to actually adding this!

Thanks again

closed 2018-12-20 by tknickman

scottbez1, 2018-12-20:
What do you think of something like this instead of using the github wiki (I just threw some of my old photos up there as a sample): https://paper.dropbox.com/doc/Split-flap-Maker-Gallery--AUAjfiV_0IiNRt23PJWrHGnRAg-ZLd5c17ZVwEDRfNgZR2n7

It does require a Dropbox account to edit (anyone can view without an account), but it took me probably 1/10th of the time it would have taken to create something similar in the wiki, so I kind of like that aspect of it.

Feel free to poke around and edit it if you'd like, I've only linked to it from this issue so far.