scottbez1/splitflap #62: Numbers only too

whitefang1997, 2019-02-19:
Hi Scott,I want to make it numbers based too.How would this affect the code and the size of the wheel hosting the cards and the entire mechanism?At splitflap-0.5.

Best wishes,
Fang

scottbez1, 2019-02-20:
Hi Fang,

Have you looked at #16 ? That issue is pretty old, and the design discussion there is somewhat outdated (there are no longer any gears), but the main issue is still relevant: spool size (and size relative to the struts that hold the spool together). The spool gets really small with just 10 flaps versus the standard 40 flaps of the alphanumeric design, violating a number of assumptions in the parametric design which results in an invalid model.

Given the fundamental issues with really small spools, I suspect your best bet would be to repeat the numbers 3 or 4 times, so the spool is either 30 or 40 flaps. I haven't looked closely at a 30 flap design though, so you'd want to make sure to carefully review the design for any potential issues.

whitebird, 2019-03-11:
My solution would probably be to set the amount of flaps to 22 and remove the odd hole, so that the resulting amount will still be 11 (10 digits + blank). In the firmware I still would have to change the amount the motor moves.

dwightmulcahy, 2019-05-10:
interested in a number only version too...

scottbez1, 2020-04-18:
Closing this in favor of #16 to track a numbers-only variant.

closed 2020-04-18 by scottbez1