scottbez1/splitflap #64: ESP32 support

whitebird, 2019-03-18:
As referenced in https://github.com/scottbez1/splitflap/issues/55 , the ESP32 is the successor for the ESP8266.

One of the issues is that there isn't a board like the Wemos D1 mini that is as ubiquitous.
An alternative is to create a board to solder the ESP32 directly, but that would require a 3.3V regulator as well as a programmer chip and a USB port.

whitebird, 2019-03-19:
To me it seems this version of the board seems mostly used: https://www.aliexpress.com/item/ESP-32S-ESP-32-Development-Board-WiFi-Wireless-Bluetooth-Antenna-Module-For-Arduino-2-4GHz-Dual/32898649271.html

scottbez1, 2020-04-28:
Starting to think through this a bit more as part of a plan to support larger displays - https://paper.dropbox.com/doc/Splitflap-large-scale-driver-project--Ay7kxGjRClsCtJ6qnOOqLQeOAQ-rUpovQ77LuERHKwqJLcEH

I'd separately like to consider an ESP32-based driver board that's still hobbyist-friendly (i.e. through-hole components), just because of the benefits of having wifi or bluetooth available for control out of the box, but for now I'm focusing on supporting larger displays.

SamWarmuth, 2020-04-28:
The combination of an ATTiny88 ($1.50, through-hole) and a TPIC6b595 (also $1.50, through-hole) for daughter boards that can each control two displays and a ESP32 running the show over I2C seems like a strategy that would work. Having a single ESP32 that drove all of the displays directly means *lots* of long wires that are pulling real amounts of current. If each of the control boards is all near its motors, you just need a single signal wire bus.

There's some more info/pictures in my previous post: https://github.com/scottbez1/splitflap/issues/66#issuecomment-549174768

I haven't had time to get any farther on the project, but I just made my repo public in case it could be useful to anyone: https://github.com/SamWarmuth/SWSplitFlap

scottbez1, 2020-04-28:
Nice, thanks for sharing - I'll have to poke around your repo next weekend. I've linked your comment and repo on the planning doc.

I don't have experience with I2C over longer distances, so I'd be somewhat worried about potential issues sharing a long I2C bus across a big display, but not sure how much of a problem that would actually be. Did you do any tests with longer cables to see what distance you can get?

My other concern would be programming that many devices in the large display - with a 120 character display I'm looking at building, that's 60 MCUs, so would probably want to have a custom bootloader so all MCUs can be programmed via the bus (and ideally can be programmed in bulk rather than one-by-one).

SamWarmuth, 2020-04-28:
I didn't test with long distances, but I had no trouble at around 6-12". The internet says up to [a few of meters](https://electronics.stackexchange.com/questions/106265/maximum-i2c-bus-length) is pretty straightforward.

Split flaps are a pretty good use-case: each PCB along the way repeats the signal, so the distances are individual, not cumulative. Also, we're transmitting really small amounts of info (e.g. "move motor 7 forward 300 pulses" is only a handful of bits depending on how you encode it), so we can use a lower bitrate with more error correction.

Programming is definitely a sticking point. With through-hole components it shouldn't be too hard to use a [ZIF socket](https://en.wikipedia.org/wiki/Zero_insertion_force) connected to a programmer to write them all. There are boot loaders you can use to [write firmware over i2c](https://hackaday.com/2018/10/20/i2c-bootloader-for-attiny85-lets-other-micros-push-firmware-updates/), but if you have to write that firmware on their in the first place, the main thing that gives you is over-the-wire updates.

scottbez1, 2021-03-25:
Closing this, as initial support for ESP32 has been implemented in firmware for a while now.

Regarding the other discussion around large displays and ESP32 support, the new [Chainlink Driver board](https://github.com/scottbez1/splitflap#chainlink-electronics) is being designed for this purpose (using shift registers and an SMD design that's relatively cheap to order mostly-assembled), along with a Chainlink Base board which will eventually host the ESP32.

Feel free to open new issues if there is any specific ESP32 functionality still missing or desired.

closed 2021-03-25 by scottbez1