scottbez1/splitflap #66: Large Display (>12 characters)

barkerb1, 2019-03-30:
Any thoughts on scaling beyond a single Arduino?  Would a Mega or Duo board allow for more controllers per micro-controller?

My initial thought would be to use a RasberryPi to daisy chain each Uno's 12 characters and also serve as a small web server for remotely controlling a large display.  I'm going to start with connecting 4 controllers (16-char) and two Uno's so I can start working on the Pi.  If successful, I plan to integrate it with my goal home and scale up to something big enough (100+ char) to display Tweets, song titles, news headlines, and stock tickers on command.

SamWarmuth, 2019-03-31:
I'm working on a build using an ESP32 and MCP23017 i2c port expanders (16 new pins per chip). You can use up to eight expanders to get 128 GPIOs. The ESP32 is a *lot* faster than an Uno, so you should be able to drive more concurrently —  not sure how many it would max out at though.

Semi-unrelated, but if you're going to be running that many characters, you might want to take a look at the TPL7407L for driving the steppers — it uses NMOS transistors instead of darlington transistors, it should run cooler (and use less power).

scottbez1, 2019-03-31:
The 2 main limits of the Uno are memory and clock cycles. A Mega won't help with the clock speed constraint so I don't think you'd be able to add many additional modules, if any. Both the Due and ESP32 are likely good options for controlling more modules, though connecting multiple Uno's to a Pi should also work fine if you've already got those handy. For a big display with 100+ characters you'd probably want to drastically rethink the electronics to make it easier to build/control and cheaper (this design is really optimized for small hobbyist production).

barkerb1, 2019-04-01:
@scottbez1, I'm not thinking of building something commercially viable, simply a one off display to hang on my wall with the added cool factor when I have guests over of being able to talk to it.  Using a Due is intriguing, if it could run at least 6 controllers (24 characters, one row) per board then I would only need 4-5 to run the whole display and would only need marginal tweeks to what you've already created.

@SamWarmuth, I'd be interested in possibly using the ESP32 if you're willing to share your fork of the project.  Honestly, I'm trying to put as little time investment into the electronics as possible so I can invest my time into the software and connecting it to the cloud, which I'm also willing to share if anyone wants it.

I just finished soldering the boards for 16 characters to begin experimenting, happy to collaborate if anyone's interested.

peazley, 2019-11-03:
@barkerb1 @SamWarmuth have either of you made progress for supporting a larger set of characters?  I'm really curious about using an ESP32 for this.

SamWarmuth, 2019-11-03:
Unfortunately I haven't had a lot of time recently to work on this in the last few months.

I ended up designing my own PCB that uses an ATTiny88 and a TPIC6b595 to drive two boards, controlled using an ESP32 over I<sup>2</sup>C. Since the ESP32 is just sending high-level commands to the ATTiny (e.g. "Move to position 13"), the CPU load is really minimal. The only issue you'd run into is the max number of I<sup>2</sup>C slaves, which is 127. But that just means you need one ESP32 for every 127 characters you want to display (or maybe come up with a more complicated I<sup>2</sup>C hierarchy). 

I also ended up redesigning the enclosure to be fully 3D-printed, to save on assembly time/effort/price (I own a printer). My estimate using this system was that I could get the price-per-display down to $6 or $7. I haven't had time to clean everything up and build more than a couple of the displays. I'll have more time to work on projects like this as it gets colder out and I spend more time inside.

Unfortunately it's not in a great place to share right now, but if there was some interest I could try to get something published.

Here are a few pictures showing the different parts:

Driver PCB
[image]

Fully 3d printed spool/frame:
[image]

With numbers:
[image]

benlongo, 2020-02-10:
Hey Sam, thank you so much for publishing this awesome project! I’d love to see the new designs you have been working on, even if they are in quite rough form.

SamWarmuth, 2020-02-11:
Splitflap is all `scottbez1`'s work, I'm working on a new build that takes a lot of inspiration from his. I'll see if I can get the files in a better state and publish them soon.

scottbez1, 2020-04-18:
Cleaning up old issues. I think the main questions have been answered here. Please comment or open a new issue if you still have outstanding questions, thanks!

closed 2020-04-18 by scottbez1