scottbez1/splitflap #67: Bigger Displays (150x100mm cards)

gollou, 2019-04-10:
I'm trying to scale up each module so they can be readable from further away, I think I'll get each A6 card by cutting some A4 sheets into 4 ([aliexpress link](https://www.aliexpress.com/item/ID-card-making-supplies-material-Blank-Inkjet-print-PVC-sheets-A4-50sets-white-color-0-76mm/32825311077.html?spm=a2g0s.8937460.0.0.34752e0eLsb6MT))
Each module should then be A5 size : 148x210mm

But I’m wondering if any of you has experimented the design with that big letters and if you can foresee any issue coming : will it be too heavy for the motor, will the wheel be big enough to resist the weight, will the 1.4mm plastic holder be resistant enough for the weight, should the depth of the box be increased ?

SamWarmuth, 2019-04-10:
I haven't tried letters that big before, but I suspect your biggest issue is going to be motor torque. The standard card flaps are about 23 cm² (4.28cm x 5.4cm), yours will be 156 cm² (10.5 x 14.9). That means each flap will be almost **seven times** heavier.

Stepper motors have more torque when they move more slowly, but I'm not sure it would ever have enough strength to move those flaps. There are more powerful motors available, but they'll cost more and will require a bit of redesigning.

scottbez1, 2020-04-18:
Cleaning up old issues. Thanks @SamWarmuth for answering this one! Motor torque is definitely a concern with larger letters - the cheap 28BYJ-48 motors are already pretty much at their limit with the current credit-card sized flaps, so you'd probably need to switch to a proper stepper motor and stepper driver to go bigger.

closed 2020-04-18 by scottbez1