scottbez1/splitflap #68: Flaps Getting Stuck (motors?)

Zuckyb, 2019-05-07:
I'm assembling two sets of split flaps and I've found that I'm having issues with the flaps getting stuck on the start-up rotations.
When the program starts tall the reels do a slow rotation and then a fast rotation. Some of the reels consistently get stuck around the same spots on the quick rotations. I was wondering if maybe the stepper motors don't have enough torque. If anyone has had similar problems and found solutions please let me know.

ericyeargan, 2019-11-20:
I had a similar issue where single-stepping through flaps worked fine but the motor would stall on long rotations. I was able to resolve it by reducing the max motor speed which I accomplished by changing the MIN_PERIOD_MICROS constant in the generate_acceleration.py script from 1200 to 1650. I arrived at that new value by trial-and-error - it's the highest number I could use without the occasional stall. I then regenerated the acceleration.h header by running that script.

Hope that helps!

scottbez1, 2020-04-18:
Cleaning up old issues. Looks like this one has been answered (thanks @ericyeargan - that's exactly the right solution!), but please comment or open a new issue if you have remaining questions, thanks!

closed 2020-04-18 by scottbez1