scottbez1/splitflap #69: About Clock

whitefang1997, 2019-05-16:
I want to make a clock, but I don't know how to rewrite Arduino Program?
I tried to add case ,but it still exists big problem.
What can I do? or something I should know?
Thank you.

Best wishes,
Fang

whitebird, 2019-07-04:
Where are you exactly gettings stuck? The clock currently is focused on being a full text display and is not easily adapted to display only numbers. More info on that here: https://github.com/scottbez1/splitflap/issues/62

scottbez1, 2020-04-18:
Cleaning up old issues. Apologies for not answering this one earlier. If you still have questions, please comment here or open a new issue, ideally with some sample code or a description of what you've tried so far that hasn't worked. Thanks!

closed 2020-04-18 by scottbez1