scottbez1/splitflap #75: Possible easier calibration?

cstarkers, 2019-11-12:
I’m just wondering if anyone has attempted to implement a simple switch as a calibration device? I’m thinking a piece of wire glued along the top front edge of a flap, and contacts on the inside of the top of the front face.  When the previous flap falls, the wired will bridge the contacts. Seems simpler and cheaper than IR/Hall effect, as it would only require a few bits of wire and 1 component (pull up or pull down resistor).

I haven’t built my display yet, but I’m planning to soon.  Unless someone can point out something I’m overlooking, I intend to attempt this and report back. 

My apologies for not attempting to draw it up, but my CAD is not up to that.

scottbez1, 2020-04-18:
Sorry for responding so late. It's an interesting idea, and I'd love to hear how it turned out if you ended up trying it. 

I'd be worried about getting calibration accurate enough that way though -- with 40 flaps there's bound to be some slight variation in when they flip down, so you generally want to make sure the motor is aligned with the middle of the "acceptable" region when calibrating. Otherwise, many of the flaps will flip correctly, but some flaps that are aligned differently or slightly larger or smaller either won't flap when they're supposed to (if the motor stops early) or will flip down before they're supposed to (if the motor moves too far for a given flap). The hall-effect sensor allows you to really finely adjust the calibration position to make sure it works for all 40 flaps.

Closing this issue for now, but I'd love to hear if you tried it and how it worked!

closed 2020-04-18 by scottbez1

SyntaxColoring, 2020-05-06:
There's an expired 1973 patent, [US3771242A](https://patents.google.com/patent/US3771242A/en), that used notches cut on the top of each flap to selectively engage an array of sensitive switches. The motor would turn until the switches matched the code it was expecting.

With software, you could probably get away with 2 switches. The first switch would be engaged only by one flap, for homing. The second switch would be engaged by every flap, to let the controller adapt to minor variations in how different flaps flip.

cstarkers, 2020-05-06:
That patent is a very interesting idea, thanks for bringing it to my attention.

I have had to put my version of this on hold for a while, but I fully intend to get back at it once Covid abates and I have laser cutter/3d printer access again - I will get back to you if I have any success.