scottbez1/splitflap #82: OSError: [Errno 2] No such file or directory

ahvinc, 2020-04-13:
Hi,
I'm trying to run generate_2d.py and i get this error 

> splitflap % 3d/generate_2d.py --panelize 60
> INFO:openscad:['openscad', '-o', 'build/laser_parts/dummy.png', '--imgsize=1,1', '-D', 'render_date="2020-04-13"', '-D', 'render_3d=false', '-D', 'render_index=0', '-D', 'render_revision="51ed870"', 'splitflap.scad']
> Traceback (most recent call last):
>   File "3d/generate_2d.py", line 56, in <module>
>     svg_output = renderer.render_svgs(panelize_quantity=args.panelize)
>   File "/Users/principal/splitflap/3d/projection_renderer.py", line 96, in render_svgs
>     num_components = int(self._get_num_components())
>   File "/Users/principal/splitflap/3d/projection_renderer.py", line 51, in _get_num_components
>     capture_output=True,
>   File "/Users/principal/splitflap/3d/openscad.py", line 90, in run
>     proc = subprocess.Popen(command, stdout=stdout_type, stderr=stderr_type)
>   File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 394, in __init__
>     errread, errwrite)
>   File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1047, in _execute_child
>     raise child_exception
> OSError: [Errno 2] No such file or directory

Any idea how to fix this ?

vagelim, 2020-04-13:
It seems you either do not have `openscad` installed or it is not available in your `$PATH`.

ahvinc, 2020-04-13:
I did install openscad so I guess as you say it is not available in my `$PATH`
I'm not so familiar with this... Any clue how to do ?

closed 2020-04-14 by ahvinc