scottbez1/splitflap #83: Use the motor driver provided with the motor ?

alexbil, 2020-04-18:
Hi,
First off all, thank you for the amazing work that yo have already done !
I did not read the whole code yet, but I was wondering if there is a possible adaptation to use the motor driver provided with the 28BUY (based on the ULN2003) ?
Thanks !

scottbez1, 2020-04-18:
Yep, you can use Basic IO mode: https://github.com/scottbez1/splitflap/wiki/Electronics#basic-prototyping-alternative-electronics-approach

However, this uses a lot more pins on the Arduino, so with an UNO you can only control 3 modules this way (instead of up to 12 with the custom driver electronics/shield).

scottbez1, 2020-04-18:
Closing this for now, let me know if you have other questions.

closed 2020-04-18 by scottbez1