scottbez1/splitflap #117: SplitFlap Display has three strange behaviors

daniele8808, 2021-01-18:
Hello Scott, It's been a long time and I would like to finish this project started more than a year ago. I have completed the construction of a module and mechanically it should be ok, I have also done some tests with the sensor and it is correctly detected but I still cannot see the letter I want on my display. In this short video I show the three problems I'm having:

https://youtu.be/BHOUhOT_epA

- 1 error: often when I select a letter to view the display freezes after 2 steps for no reason
- 2 error: if I try to display a character that is too far from the starting point (for example I switch from A to Z), the display makes a complete revolution and returns to A, but no errors are reported on the console
- 3 error: a few times when I recalibrate by entering the @ the display arrives at the home point and adds a micro step that changes the initial letter

I would be really grateful if you could help me complete this project.

dmadison, 2021-01-18:
It looks as though the display is binding at 'A'. [Look closely around 38 seconds](https://www.youtube.com/watch?v=BHOUhOT_epA%3Ft%3D&t=37s) - it holds the 'A' position for ~2 seconds before it relaxes, which indicates that it's trying to move the spool but can't. I've noticed a similar phenomenon on my own test modules when the spool gets out of position.

Scott may have better advice, but I'd start by checking that the flaps aren't hitting the enclosure anywhere except at the very top edge.

daniele8808, 2021-01-20:
thanks for the advice, I try to cut the displays again using the v.6 but if the problem was the beat of the cards it would solve only one of the three problems I am having. Let's say that the main problem I'm having concerns point 1. I still can't get the letter I want to display, it works randomly and resets itself continuously.

scottbez1, 2021-01-21:
My guess is that the motor is stalling/skipping due to not having enough torque at the higher speed. The behavior you're seeing is that the module will always rotate slowly (which has more torque) when it is searching for the home position at startup (or if it detects an error while moving it will also search for the home position before continuing), and then once it's found it will return to normal mode and accelerate up to full speed to go to the desired flap. It seems like everything works fine until it starts accelerating, so the first thing I would try is to reduce the speed - the default values are fairly aggressive and may not work if there is more friction, or you have a slightly different motor, etc.

You can reduce the top speed by closing the Arduino IDE, then edit generate_acceleration.py (should be in the same folder as the arduino code) to change the MIN_PERIOD_MICROS value - maybe try setting it to 2400 to start, which should be half the default speed. Then run that script (should be directly executable on Mac or Linux, or you may need to invoke it via python from the command prompt on Windows), which will generate a new acceleration curve in acceleration.h. After that you can re-open the Arduino IDE and upload the code again and test it. If that works you can try some lowering the value again to find the fastest speed your module will allow reliably.

daniele8808, 2021-01-25:
Hi Scott, thanks for your reply and time.
I've done what you told me but I think I have two problem at the moment:

1. mechanical problem
2. connection/software problem

- Regarding the first problem I tried to make the change via code you told me by changing the value in the python file but without success. In the end I decided to cut the new version of the project the V 6.0 where I saw you moved the motor by removing the two gears (so as to eliminate the friction problems I had) and I will also buy the 12 Volt motors (more powerful).
I hope to cut the new part in this week and make a new test.

- Regarding the second problem the few times that the gears turn well and I have no problems with attractiveness, however, I find it difficult to visualize the required character and I don't understand why. As you can see from this second video some times I specify the character that I would like to display but it is not printed in the console, instead I display an empty string. Does it happen to you too or do you know why?

here a video of this second problem: https://www.youtube.com/watch?v=e1L3PMftkNk&feature=youtu.be&ab_channel=DanieleMartini

dmadison, 2021-01-25:
For the second problem, in your video you're typing spaces between `=` and the character you want to send. The firmware is interpreting that as you want to write the space character to the first module (and the character you typed to the second). You have to send the command without any spaces.

daniele8808, 2021-02-02:
Hello Scott, I finally finished my display but I need help for the last time. Here the display -> https://youtu.be/7kNUpaLQhlQ
I read your guide about calibration but I can't display what I want.
All the sensor looks work.
What do you think is the problem?
Thanks in advance for any help

Daniele

daniele8808, 2021-02-05:
Ok I think I foud the solution By myself. The cards I cut are not very accurate so this creates small malfunctions in the system, but all in all it seems to me to be working. This weekend I try to make new cards and improve the functioning of my display. In the meantime, I made this little video to thank Scott for This beautiful project.

https://www.youtube.com/watch?v=u_ax8iPwcRU&feature=youtu.be&ab_channel=DanieleMartini
https://spinzam.com/shot/?idx=254804

I have only one last question, in the latest version 6 code I saw that a piece of code concerns a bluethoot connection with the. ESP32 module. Is there any documentation on how to integrate this module as well?

Thanks in advance for everithing
Daniele

scottbez1, 2021-03-05:
Daniele, I just noticed when watching your latest video - I think the front faces on your modules are installed upside down/inside out.

The taller portion should be on the top, and the shorter portion on the bottom.
[image] [image]

If you flip the front face vertically (so the side that was facing in now faces out) I think that may help with reliability, as it will hold the flaps back more consistently at the top. See https://youtu.be/UAQJJAQSg_g?t=36 for a demonstration of why that position is key to reliable behavior.

(it's unfortunate that the design is subtly asymmetric like this, but @dmadison has smartly [added engraved orientation markers](https://github.com/scottbez1/splitflap/pull/111) in an upcoming revision of the hardware design so it's hopefully easier to notice in the future)

daniele8808, 2021-03-13:
............. :-) ok thank you Scott. I'll fix it and see if my display start to work better making this change.
Meanwhile I already started to build my second display. I'll pay more attention during the assembling process. 
Thanks

closed 2021-03-13 by daniele8808