scottbez1/splitflap #127: 3D Printer?

ezorita, 2021-02-28:
Hi Scott, this is a fantastic project. Thanks for your contribution.

Not sure if this has been asked before. Is there an easy way to export the laser cut model to 3d printed parts?

lastcoolnameleft, 2021-03-05:
@ezorita I have been working on exactly this.  It is possible; however, there's some gotchas that I haven't quite solved yet.  My spool doesn't seem to spin as loose as the laser cut ones.  I'm sure I could change some of the dimensions, but it's quite time consuming.  I've modified his splitflap.scad so that I can 3d print the flaps, spool and frame.  Let me know if you're interested and I can share with you my findings so far.  

And yes, this is an exciting project and I think some others have tried 3d printing it, but I haven't seen any final product of it yet.  I feel like I'm close, but the tolerances are really low.

ezorita, 2021-03-08:
hey @lastcoolnameleft it's great to hear you already made some progress on this! I did start to adapt the different frame parts to 3d printable objects.

It would be great if we share our progress, I'll try to improve upon the printing tolerances. Would you like to fork the repo so that we got a place to contribute and eventually open a PR back to @scottbez1?

dmadison, 2021-03-08:
For what it's worth the flat laser-cut design should work (nearly) out of the box for 3D printing even though it's not designed for it. Within the `splitflap.scad` file you need to modify:

```
render_3d = false;
kerf_width = 0.5;  // spacing between parts
```
And you'll get the entire 3D design packed flat which you can export as an STL for printing.

The one exception is that one of the spool struts is placed on top of the right side panel so it's cut out of it on the laser. You'd have to reposition the strut somewhere else.

scottbez1, 2021-03-09:
Yep, thanks for the summary, Dave!

By the way, a lot of the infrastructure needed to formally support 3d-printable exports should be there already; if someone wanted to create a script for it and send a PR that'd be really sweet!

I think it would just require a small tweak to projection_renderer.py to pass in the file extension to use when rendering each individual component (.stl instead of .svg), and then would need a new wrapper script akin to generate_2d.py (it would be simpler though, since you don't need to do any of the post-processing or merging that's done for the SVG files). Since that exports each component individually already, the spacing/layout becomes irrelevant and you can lay out the individual STL files yourself using your plater/slicer of choice.

The only thing that I think would be kind of weird is the engravings, which are exported separately from the body they're engraved onto. That approach wouldn't play nice with an STL export, but the current engravings are also probably too small to be worthwhile on a 3d print, so it may be best to just skip engravings initially to keep things simple.

lastcoolnameleft, 2021-03-11:
Instead of having 3d printed parts like laser-cut parts, I wanted to build an integrated solution that let you toggle flags to render the specific piece you want. (e.g. instead of 4 walls for the frame, render the full frame)

I hope to have it better documented, but here's my attempt at a 3d printed model:
https://github.com/lastcoolnameleft/splitflap/blob/3d-print/3d/splitflap.scad

The changes I made are to add a number of "render_xxx" flags to make it easier to export combined pieces instead of individual parts.  For example setting `render_spool` and `render_spool_motor_wheel` to true will print one side of the spool as one piece and `render_spool_struts` as the other.

I also added `render_frame` and `render_captive_nuts` flags to render the frame as 1 assembled piece.  

The issue I hit was that it just added to the code complexity and assembling everything became a lot trickier.  One idea was to have individual pieces split up into separate files similar to how there's a pcb.scad which would make the code more composable.  Then there could be a "laser_cut.scad", "3d_print_frame.scad", etc.

I'm definitely open to ideas.

dmadison, 2021-03-12:
How divergent are you imagining the 3D printable design to be?

If it only differs mildly from the laser-cut design (e.g. hiding captive nuts) then a few flags might be fine, coupled with its own module for all of the parts in 3D-printable orientation. Anything more than that and it might be worth splitting the 3D printed version into its own design/file just to keep things from getting too disorganized. It would probably make more sense to rework the `splitflap.scad` file so that the 2D render is conditional; that way it can be `include`'d in other designs and you can reference off of all variables without having to expose them as functions.

I do think the separate flags for each part is overkill. A script could be used to automate exporting the individual parts, probably by iterating through child objects like the projection renderer.

lastcoolnameleft, 2021-03-12:
Great question.  I initially thought it wouldn't be very divergent.  However, I'm hitting concerns that I don't have an answer to (e.g. If the frame is all 1 piece, how do you fit the spool + gear into the frame.  Currently, I brute force it.)

I'm still learning the repo and poked around the /scripts and it appears that openscad.run() lets you override existing variables to generate STL's.  Is that accurate?

I looked into creating a `render_for_3d_printing()` function which sets all of the appropriate variables and overrides to generate the appropriate pieces, but OpenSCAD variable scoping didn't seem to allow for that.  This is my first OpenSCAD project, so let me know if there's an option I've missed.

scottbez1, 2021-03-12:
Yeah, the python scripts are able to override variables using the `-D` command line option (https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment ), whereas that kind of behavior won't be possible from within another scad file without first refactoring splitflap.scad to wrap more things in high level modules. The global variable situation is not great (but in my defense this was also _my_ first OpenSCAD project :stuck_out_tongue: so I didn't have a great sense of OpenSCAD architectural patterns). It's definitely in need of some refactoring. Unfortunately I can't commit to that myself at the moment, but if someone does take on that heroic task I'd very much appreciate if it's done in small increments and separate from any functionality changes for the sake of reviewability.

That said, like @dmadison mentioned, it's probably worth considering the end goal before getting too deep in the splitflap.scad weeds. The design as it stands today was very much built for laser-cutting, with that assumption baked into nearly every design decision. At a certain point it may make sense to just factor out some of the core dimension calculations (spool dimensions, window size and position relative to the motor, etc) to a library and build the actual pieces up from scratch using just that core library.

The best example of "built for laser-cutting" is probably the spool, which is currently an assembly of 7 pieces plus a bolt and a nut, when it could easily be just 2 pieces total (or 1, depending how well your printer can handle supports) if 3d printed. If you're interested in making a variant that's optimized for 3d printing, rather than just 3d printing the existing design, consider this a gentle nudge towards starting there :wink: , as I think it's the biggest opportunity to really leverage the benefits of 3d printing over laser-cutting. For instance, it should enable much smaller diameter spools, which would solve the most common question of "how do I build a number-only display" without needing to increase the gap between flaps substantially (https://github.com/scottbez1/splitflap/issues/16#issuecomment-553883245 )

sohowaty, 2021-03-12:
Just to add my two cents to 3d printing - I wouldn't want to print the whole frame as-is for laser cutting - it'll require a lot of time to print when compared to simplified version. NilSkilz made a fork with models designed for 3d printers - two-part spool and one-part frame. This fork is old and I'm not sure if all dimensions are still valid when copared to 'master' version, but I printed one module and going to test it ASAP.

lastcoolnameleft, 2021-03-23:
I just took a first-stab at refactoring `splitflap.scad` so that it would be more easily 3d-printable.  I'm interested in your feedback.

KostasDaskalos, 2021-04-18:
As I type this comment I am printing on my Creality Ender 3 V2 a 3D model I made from the laser cut SVG. I imported the SVG to CorelDraw and separated all the individual shapes closing all paths. I then exported the file as Auto CAD DXF and then I imported it to Fusion 360. I used DXF because I found out that preserves the actual measurements of the original shapes. I am new to Fusion 360 and I do not know how to properly import SVG files... I extruded all shapes to 3,1 mm height and export them as STL files. I know that maybe there is a way a faster and better way to do it but until now it seems it is exactly what I needed!
Here is a picture of the model printing:
[image]
I will assemble it and see what I did wrong, modify the model and then post the final correct one here! I will update when I have more information!

dmadison, 2021-04-18:
That looks great @KostasDaskalos! If you do need to make any changes, there is a way to export the STLs directly from OpenSCAD. [See my comment earlier in the thread](https://github.com/scottbez1/splitflap/issues/127#issuecomment-792819989).

KostasDaskalos, 2021-04-18:
The only 3D CAD program I know is Fusion 360 ( and I am still learning... ). I am old enough to get into another one ( OpenSCAD ), my brain can not hold that much information any more... Importing STL files to Fusion 360 and working with them is a bit troublesome so I did ...all these steps to import DXF to Fusion 360. Now I can change anything I want and export STL files from Fusion. Tomorrow I will have all pieces printed and assembled ( still needs 5 more time printing... ).
At some point in the future I may take a look at OpenSCAD...

KostasDaskalos, 2021-04-19:
A quick update, as I am still 3D printing one side part: 
[image]
It needs a bit ...sanding around the corners where the pieces press fit but over all I think it is a nice model. It is as well so nice press fitted that actually it can be build without screws!
One piece left. I will test it and I will post the final STL result here!
Have a nice day :)

KostasDaskalos, 2021-04-19:
Done!
Proof Of Concept:
https://www.youtube.com/watch?v=AqFgEZLnIoI
I am sorry for the bad video quality but I do not have a very good phone ... Link for the files is in the description of the video but here it is again:
https://drive.google.com/file/d/1yVeEgPqHvn44Lm4M6I_Hzljlyx13a50h/view?usp=sharing
Some notes regarding the final models:

- Yes some tweaking could be done better so the models do not need sanding but at least the fit is tight! I am pretty sure it can work without screws!
- The body of the spool can be done in one piece with minimal support but I will do it in a future version.
- The holes in the spool wheels seem to be very fragile but they are not! Tried to break one but it is sturdy enough!

Creality Ender 3 V2 slicer settings:

- Layer Height: 1,2
- Extruder temperature: 200
- Bed temperature: 60
- Filament: PLA 1,75

Have fun with it! And **thank you** for the **awesome** design!

Goopinson, 2021-04-21:
Just wondering, does the spool spin smooth enough? Also, were you able to print multiple and connect them as designed?

KostasDaskalos, 2021-04-21:
I am currently in the process of making the electronics. I tried spinning the spool without cards with an Arduino and some hacked up circuit on breadboard and it works fine! I am going to print 3 more and do a 4 digit display. When I do it I will update here with images and a better ( hopefully... ) video!

len0rd, 2023-02-24:
for those coming across this thread looking for a more printer-friendly module case, i ended up consolidating the v0.7 components from 12 pieces to 4 to streamline the print. I used Fusion360 to combine the components since I dont have the time to invest in learning openscad. After a couple iterations everything is working well for me: Here are [the resulting stls](https://www.printables.com/model/407557-split-flap-display-module-case)

[image]

Ended up down this rabbit hole after accidentally breaking one of my laser-cut pieces. That was easy-enough to replace independently thanks to the easy stl generation scripts! Thanks for the great project

chip-davis, 2023-12-21:
> for those coming across this thread looking for a more printer-friendly module case, i ended up consolidating the v0.7 components from 12 pieces to 4 to streamline the print. I used Fusion360 to [...]

This looks great! Any plans to incorporate the inter-connecting? Thanks!

len0rd, 2023-12-21:
@chip-davis no plans at the moment. I’ll take a second look but I didn’t end up needing them for my project. I made a 36 module display and just used the captive front screws to secure all the modules to a piece of ply board. That was rigid enough for me.

[image]

JHthe4, 2024-10-31:
Hi everyone,

what a great project this is, and a nice DIY alternative to a ready-made solution like the Vestaboard! I'm price-sensitive, so I was wondering if a clear winner has emerged yet between laser cutting the parts and just 3D printing them, and how far this project is with supporting 3D printing out of the box (for enclosures and the flaps themselves), compared to alternatives like https://github.com/dmlambo/SplitFlap . In particular, I like that this project's code and addressing system seems to be much more mature, but I'm wondering about cost differences when looking at a ~60 module display.

Thanks in advance for any insights you can give!