scottbez1/splitflap #128: Flaps going to wrong letters

lastcoolnameleft, 2021-03-05:
I've noticed an issue that I've worked really hard to pinpoint and I think might be software related:

To reproduce the issue:
With only 2 modules, in the console type:
```
@
aa
by
yb
```

Expectation:
Both split flaps will go through the full cycle, then to go `aa` then `by` then `yb`.

What happens:
Both split flaps will go through the full cycle, then to go `aa` then `by` then `rb`.

The logs indicate that it should be on `yb`, but the module stops moving past `r`.  The funny thing is if I do `@` after those commands, it goes to `yb`, where I would expect.  I have consistently reproduced this problem.

The reason I don't think this is a mechanical problem is because I swapped which ports the modules were plugged into and was able to reproduce the problem. (e.g. If it was a mechanical problem, then I would have expected the same module to have a problem, but instead it was always module 1).

Here's the debug logs:

```
@VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=3
Going to flap index 3. Current step is 0. Delta is 154
VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=3
Going to flap index 3. Current step is 0. Delta is 154
=aa\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=38
Going to flap index 1. Current step is 154. Delta is 1946
aVERBOSE: DELTA FLAPS=38
Going to flap index 1. Current step is 154. Delta is 1946
a"}
VERBOSE: Found expected home.VERBOSE: DELTA FLAPS=40
Calculated new expected home 4096.
OLD:us=255, ue=2036, m=2060
NEW:us=2307, ue=4084, m=4108
VERBOSE: Found expected home.VERBOSE: DELTA FLAPS=40
Calculated new expected home 4096.
OLD:us=255, ue=2036, m=2060
NEW:us=2308, ue=4084, m=4108
{"type":"status", "modules":[{"state":"normal", "flap":"a", "count_missed_home":0, "count_unexpected_home":0}, {"state":"normal", "flap":"a", "count_missed_home":0, "count_unexpected_home":0}]}
=by\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=1
Going to flap index 2. Current step is 2100. Delta is 51
bVERBOSE: DELTA FLAPS=24
Going to flap index 25. Current step is 2100. Delta is 1228
y"}
{"type":"status", "modules":[{"state":"normal", "flap":"b", "count_missed_home":0, "count_unexpected_home":0}, {"state":"normal", "flap":"y", "count_missed_home":0, "count_unexpected_home":0}]}
=yb\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=23
Going to flap index 25. Current step is 2151. Delta is 1177
yVERBOSE: DELTA FLAPS=17
Going to flap index 2. Current step is 3328. Delta is 871
b"}
VERBOSE: Found expected home.VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=2307, ue=4084, m=12
NEW:us=260, ue=2036, m=2060
{"type":"status", "modules":[{"state":"normal", "flap":"y", "count_missed_home":0, "count_unexpected_home":0}, {"state":"normal", "flap":"b", "count_missed_home":0, "count_unexpected_home":0}]}
```

I'm interested to know if you're able to reproduce this problem.  Let me know if there's any additional information I can provide to help debug.

scottbez1, 2021-03-05:
After the `=yb` command that ends up at the wrong position, can you send the same `=yb` command again and tell me what the behavior is?

Also, just for my info, what electronics hardware are you using to drive the motors? And if you can, what is the resistance measured between the red and blue wires on the motor connector? (you may have to wait a little while for the measurement to settle after connecting a multimeter)

lastcoolnameleft, 2021-03-05:
If I send the `=yb` command again after the first one, the behavior is that the first modules appears to lose home and then correctly displays `yb`

I am using an Arduino Uno with the board purchased from your Tindie store with the Commit: a1eb737 Date: 2019-01-14 (i.e. no deviation from your ordering guide)

All of my motors are show between 105-108 ohms when I measure between the red and blue wires (I'm colorblind, so I assume you mean the 1st and last wires.  Let me know if I got that incorrect)

Here's the logs from the start:
```
@VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=25
Going to flap index 25. Current step is 0. Delta is 1280
VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=2
Going to flap index 2. Current step is 0. Delta is 103
=aa\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=16
Going to flap index 1. Current step is 1280. Delta is 820
aVERBOSE: DELTA FLAPS=39
Going to flap index 1. Current step is 103. Delta is 1997
a"}
VERBOSE: Found expected home.VERBOSE: DELTA FLAPS=40
Calculated new expected home 4096.
OLD:us=255, ue=2036, m=2060
NEW:us=2308, ue=4084, m=4108
VERBOSE: Missed expected home! At 2060. Expected between 2036 and 2060.
VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=1
Going to flap index 1. Current step is 0. Delta is 52
{"type":"status", "modules":[{"state":"normal", "flap":"a", "count_missed_home":0, "count_unexpected_home":0}, {"state":"normal", "flap":"a", "count_missed_home":1, "count_unexpected_home":0}]}
=by\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=1
Going to flap index 2. Current step is 2100. Delta is 51
bVERBOSE: DELTA FLAPS=24
Going to flap index 25. Current step is 52. Delta is 1228
y"}
{"type":"status", "modules":[{"state":"normal", "flap":"b", "count_missed_home":0, "count_unexpected_home":0}, {"state":"normal", "flap":"y", "count_missed_home":1, "count_unexpected_home":0}]}
=yb\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=23
Going to flap index 25. Current step is 2151. Delta is 1177
yVERBOSE: DELTA FLAPS=17
Going to flap index 2. Current step is 1280. Delta is 871
b"}
VERBOSE: Found expected home.VERBOSE: DELTA FLAPS=40
Calculated new expected home 4096.
OLD:us=255, ue=2036, m=2060
NEW:us=2308, ue=4084, m=4108
{"type":"status", "modules":[{"state":"normal", "flap":"y", "count_missed_home":0, "count_unexpected_home":0}, {"state":"normal", "flap":"b", "count_missed_home":1, "count_unexpected_home":0}]}
=yb\n
{"type":"move_echo", "dest":"VERBOSE: DELTA FLAPS=40
Going to flap index 25. Current step is 3328. Delta is 2048
yVERBOSE: DELTA FLAPS=40
Going to flap index 2. Current step is 2151. Delta is 2048
b"}
VERBOSE: Missed expected home! At 12. Expected between 4084 and 12.
VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=25
Going to flap index 25. Current step is 0. Delta is 1280
VERBOSE: Missed expected home! At 12. Expected between 4084 and 12.
VERBOSE: Found home!
VERBOSE: DELTA FLAPS=40
Calculated new expected home 2048.
OLD:us=0, ue=0, m=0
NEW:us=255, ue=2036, m=2060
VERBOSE: DELTA FLAPS=2
Going to flap index 2. Current step is 0. Delta is 103
{"type":"status", "modules":[{"state":"normal", "flap":"y", "count_missed_home":1, "count_unexpected_home":0}, {"state":"normal", "flap":"b", "count_missed_home":2, "count_unexpected_home":0}]}
```

scottbez1, 2021-03-06:
I'm not able to reproduce this.

I would start debugging by setting `HOME_CALIBRATION_ENABLED` to false in config.h https://github.com/scottbez1/splitflap/blob/095603151c19d6603472c50c959f4c5708cc02f1/arduino/splitflap/Splitflap/config.h#L18-L21

This will eliminate any potential sensor issues influencing the behavior.

Now you can just repeatedly send `=  \n` (equal, space, space, newline) and each time both modules should complete a full revolution, ending on the same letter they were on when the Arduino started. Do this a bunch of times (at least 10 in a row)

There are two things you might notice:
- if a module stops spinning very early after just one or two rotations (and hums/vibrates while stopped until the other module stops moving too), this suggests a mechanical/torque limit. If that is the case, the best solution is probably to slow down the max speed (since torque decreases at higher speeds). You can do this by modifying generate_acceleration.py to set a higher minimum period (maybe 1500 instead of the default 1200). https://github.com/scottbez1/splitflap/blob/095603151c19d6603472c50c959f4c5708cc02f1/arduino/splitflap/Splitflap/src/generate_acceleration.py#L20 Then run that script, which updates acceleration.h, and then re-upload the code to the Arduino (if using the Arduino IDE, make sure to close it while running the script and then reopen it)
- The other possibility is that after many rotations you notice the module is slowly drifting from the correct location, most likely rotating slightly too far each time. This does not sound like it's the case for you, but this would indicate that the motor has a different gear ratio than the expected 64:1. There are some comments about changing the gear ratio in splitflap_module.h - https://github.com/scottbez1/splitflap/blob/095603151c19d6603472c50c959f4c5708cc02f1/arduino/splitflap/Splitflap/src/splitflap_module.h#L35-L40

If something else happens, can you record a video of the test and link to it here? Thanks

scottbez1, 2021-03-21:
Any luck tracking down the issue @lastcoolnameleft ?

lastcoolnameleft, 2021-03-23:
Unfortunately no.  I think that even though I put in extra due diligence to make sure it wasn't an issue on my side, that it ultimately was PEBKAC.  I haven't spent much time with it as I'm anxiously awaiting the larger controller board to spend my next chunk of time on.

I'll close for now and re-open in case I encounter the issue again.  Thanks for your investigation.

closed 2021-03-23 by lastcoolnameleft