scottbez1/splitflap #129: Request to release Ponoko SVG with etched indicator

lastcoolnameleft, 2021-03-08:
The Ponoko SVG does not include etched indicator from #102

https://github.com/scottbez1/splitflap/releases/download/v0.6/v0.6_laser_vector_ponoko_mdf_4x.svg

This is a minor change and doesn't require a full release.  Just an FYI for the next release.

scottbez1, 2021-03-09:
Yeah, there have been a bunch of small improvements to the laser-cut designs in the last few months that are not yet released. I'm planning to order some test pieces from Ponoko soon (within the next month or so) to validate them, and assuming that goes well, will create a new release at that time.

scottbez1, 2021-05-13:
Whoops, it's been 2 months, but I've gone ahead and placed a Ponoko order in acrylic with the latest design from `master` (23114e3) to validate and will promote that to a full release once I've received and tested it.

scottbez1, 2021-07-02:
v0.7 is tagged, artifacts posted, and ordering document updated: https://github.com/scottbez1/splitflap/releases/tag/v0.7

closed 2021-07-02 by scottbez1