scottbez1/splitflap #153: 3D printed flaps

wifasoi, 2021-05-25:
I think it's something that can automate the pain of sourcing, cutting and sticking the letter for making each flap.
For multimaterial printing you need to separate each color/material in a separate STL/mesh file.

For doing si I hacked the openascad script for generating the multi material file for me, and now I'm printing the flaps with my 3D printer; the result are: good enough :D

The question if this is something you want to support, so I can put some effort to clean up the script to support this output (maybe using something based on #113).

scottbez1, 2021-06-24:
Hi @wifasoi, sorry for taking a while to respond! 

I'm happy to include this and support it, so long as it can be done without _too_ much deviation from existing design files, so that the cognitive load and maintenance burden is minimized. I'm assuming that mult-material flap printing will be relatively rare (though [this reddit post](https://www.reddit.com/r/3Dprinting/comments/o4boxk/i_designed_my_own_splitflap_display/) has me rethinking that assumption), so the complexity it adds needs to be very low in order to be worth the inclusion.

That said, I currently have another flap refactor -- an alternative to #113 -- [in progress (#154)](https://github.com/scottbez1/splitflap/pull/154), which may make it tricky/painful to modify flaps for 3d printing until that has landed - the flap 3d model has been moved to [its own file](https://github.com/scottbez1/splitflap/pull/154/files#diff-0797a2e64dbec67d5df94bc8d6f85b8fbd367e0b2ec6d6c6568330ed0a6e8b28R157) and now incorporates the letter positioning as part of the module as well - maybe a useful step towards supporting this, or maybe it complicates things?

By the way, did you take a look at `generate_stl.py` and `ColoredStlExporter`? Those may be useful for the export process - it generates separate STL files automatically per color and is used to build the 3d model seen [here](https://scottbez1.github.io/splitflap/) - but I'm curious if you came up with a different approach.

caseydierking, 2021-07-15:
Scott, can the files exported from the generate_stl.py be 3dprintable to work with the overall design?

wifasoi, 2021-08-30:
Sorry for the super late reply.
So i rewritten my code based on #154, and the flap refactor helped alot :D
So a description of what I did to archive my goal:
* I made the letter thicknes a module parameter, and sunk it down in the flap if the tickness value is < 0 (https://github.com/wifasoi/splitflap/commit/74833251a91c6dc9595a0991dfbaec15bc914aa9)
* create a new .scad file that implement two module for getting the letters and the flap (for now just hardcoded with a subtract with the letters) (https://github.com/wifasoi/splitflap/commit/c018901470745a8cef6406d3b3f529cbb07066a5)

What is missing:
* modify (maybe `generate_fonts.py`) or create a python script to export the models.

Anyway, is this an acceptable contribution I can open a PR and we can discuss further.

scottbez1, 2021-08-31:
Cool, yeah those 2 commits look good! Would love a PR to add those