scottbez1/splitflap #177: Unable to run generate_font.py

TheoLong, 2021-09-14:
Here is the command I'm running:
python scripts/generate_fonts.py -t '0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ,.?-/=$%()' --ncolumns 10

I'm running this at the 3d directory.

Here's the traceback:

> Traceback (most recent call last):
>   File "scripts/generate_fonts.py", line 27, in <module>
>     from svg_processor import SvgProcessor
>   File "/Users/booker/Desktop/splitflap/3d/scripts/svg_processor.py", line 342
>     width_node.setAttribute('d', f'M 0 10 l 0 5 l {width_mm} 0 l 0 -5')
>                                                                      ^
> SyntaxError: invalid syntax

I'm running this on Mac OS BigSur v11.1

scottbez1, 2021-09-14:
That looks like a python version issue; you'll need to use python 3 or above with most python scripts in the repo, as python 2 is end-of-life. You should be able to execute the file directly (i.e. `./generate_fonts.py`) which will use python 3, or if you are are running the script by invoking python you'd need to use a python 3 interpreter (generally available as `python3` if it's installed).

scottbez1, 2021-09-17:
Just checking in - did this work?

scottbez1, 2021-09-21:
Closing due to no response, since this seems likely to be a python environment issue not a code issue. Please comment/reopen if using python3 does not resolve the issue.

closed 2021-09-21 by scottbez1