scottbez1/splitflap #178: Can't run generate_2d script

wussuphaters, 2021-09-16:
Hello ! I cannot seem to run the python scripts even though I have installed openScad, please help ! 

```
Traceback (most recent call last):
  File "/Users/maelponchant/Documents/Découpe laser/Walid Shalby/splitflap/3d/scripts/openscad.py", line 95, in run
    proc = subprocess.Popen(command, stdout=stdout_type, stderr=stderr_type)
  File "/usr/local/Cellar/python@3.9/3.9.2_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/Cellar/python@3.9/3.9.2_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'openscad'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/maelponchant/Documents/Découpe laser/Walid Shalby/splitflap/3d/scripts/generate_2d.py", line 114, in <module>
    svg_output = renderer.render_svgs(panelize_quantity=args.panelize)
  File "/Users/maelponchant/Documents/Découpe laser/Walid Shalby/splitflap/3d/scripts/projection_renderer.py", line 108, in render_svgs
    num_components = int(self._get_num_components())
  File "/Users/maelponchant/Documents/Découpe laser/Walid Shalby/splitflap/3d/scripts/projection_renderer.py", line 47, in _get_num_components
    stdout, stderr = openscad.run(
  File "/Users/maelponchant/Documents/Découpe laser/Walid Shalby/splitflap/3d/scripts/openscad.py", line 100, in run
    raise RuntimeError('Could not find openscad! Make sure you\'ve installed OpenSCAD and it\'s on your PATH '
RuntimeError: Could not find openscad! Make sure you've installed OpenSCAD and it's on your PATH environment variable
```

scottbez1, 2021-09-17:
Hi @wussuphaters. Unfortunately the scripts only support linux for now. Mac support shouldn't be _too_ hard to add, and there was a pull request to add Mac support a few years ago (https://github.com/scottbez1/splitflap/pull/13/files) but it was never completed by the author and I have not yet revisited it myself since I do all development on linux. You could try applying some of the changes from that PR to get it working (and if you do, a new PR would be greatly appreciated) otherwise I will probably not have time to revisit Mac support for another month or so.

icep87, 2021-09-23:
@wussuphaters was it openscad.py that you had issues running on Mac OS X?

scottbez1, 2023-10-08:
Closing out some old Issues. I believe this has been resolved as of #199

closed 2023-10-08 by scottbez1