scottbez1/splitflap #194: Help

DJboyy1, 2022-01-11:
Can i ask some help for the python programs, i tried a lot and still nothing for the SVG file

AirmanEpic, 2022-01-11:
I'll see what I can do, others can feel free to also take a look, it might take me a while

DJboyy1, 2022-01-18:
Normaly when you run the python file it will give you a SVG file for the laser cutter but when i try the program it runs but it give no file back.

scottbez1, 2023-10-08:
Sorry for not responding earlier. The SVG exports for laser-cutting are currently broken in newer versions of python, which could be what you were running into. In any case, I'm closing out some old issues, but please open a new issue with as much debugging context as possible if you still have questions, thanks!

closed 2023-10-08 by scottbez1