scottbez1/splitflap #198: Chainlink: Magnet check

gorwah, 2022-02-11:
If I’m making a chainlink system (no Arduino), then how do I check that the magnet is in the right direction?

mattsains, 2022-03-24:
I have the same question. Looking at the code, I think you can set SENSOR_TEST to true in `arduino/splitflap/Splitflap/config.h` and upload to the device using platformio and the chainlink env. You connect the sensor to the chainlink board. The esp32 will output the state of the first five sensor ports over serial.

I'm unable to test this because my chainlink boards have not arrived.

gilles-yvetot, 2022-07-18:
@mattsains that's what I have done and the program runs, I think it prints 0 when the magnet is not detected. I have not been able to make it display 1 though

scottbez1, 2023-10-08:
Setting SENSOR_TEST should work as https://github.com/scottbez1/splitflap/issues/198#issuecomment-1078515102 mentioned, or an easier approach would be to send `%` via serial to toggle between sensor test mode and normal mode.

closed 2023-10-08 by scottbez1