scottbez1/splitflap #209: Arduino Not Working

josecardona7, 2022-05-23:
Hi All, I have a 6 module set up that works fine when I use PlatformIO. I am really struggling to program the functionality I want since I can't figure out the PIO environment so am hoping to just use the Arduino IDE. The code directly downloaded from github, unedited (except to say 6 modules), and uploaded to the board does not work right. The first 3 splitflaps work, the 4th one rotates but misses the sensor, even though it says "normal" in the serial output, the 5th and 6th never rotate and give a "sensor error" in serial. 

I can send messages and the first 3 flaps work fine. I know it's nothing mechanical since it all works on PIO. I know the documentation says limited support for Arduino IDE but it mentioned it said to open a git issue if I saw any issues so here I am.

Thanks!

josecardona7, 2022-05-23:
Also wanted to add, when NUM_MODULES is higher than 6 (I tested 12), all 6 modules turn, but again only the first 3 work properly, the other 3 give a sensor error.

scottbez1, 2023-10-08:
Apologies for not responding before. I'm closing some old issues; if you are still having issues, please reach out in the Discord server or open a new issue, thanks!

closed 2023-10-08 by scottbez1