scottbez1/splitflap #215: task_wdt: Aborting

manusovich, 2023-01-02:
TTGO loads only once out of 10-20 times. For the rest it gives the following error:

```
{"type":"init", "num_modules":5}
E (10249) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10249) task_wdt:  - Splitflap (CPU 1)
E (10249) task_wdt: Tasks currently running:
E (10249) task_wdt: CPU 0: IDLE0
E (10249) task_wdt: CPU 1: Splitflap
E (10249) task_wdt: Aborting.
abort() was called at PC 0x400e78cf on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400862b0:0x3ffbe530 0x40086525:0x3ffbe550 0x400e78cf:0x3ffbe570 0x40084c01:0x3ffbe590 0x400f4a6b:0x3ffbc4b0 0x400e7ee3:0x3ffbc4d0 0x4008962a:0x3ffbc4f0 0x40088195:0x3ffbc510

Rebooting...
```

scottbez1, 2023-01-02:
Can you also provide the git commit hash and which env you used to upload the code (e.g. `esp32`, `chainlink`, `chainlinkBase`, etc)? And to confirm, this is on a TTGO T-Display ESP32? Thanks!

manusovich, 2023-01-03:
Hi @scottbez1 , 

Thank you for quick response. I used `chainlink` and most recent code from git (96cd98a8658db12edfb76b8447c0e9a83189b0e2) 

BTW, I don't have this issue anymore. I am not sure what happened but now I everything works well. Just in case it will help someone in the future, I did the following:

1. Checked out most recent code from git 
2. Compiled & flashed

--- At this point I experienced the issue ----

3. Installed ESP32 and TFT lib inide Arduino  
4. Flashed a few examples from Arduino TFT lib into TTGO Display. Everything was ok 
5. Returned back to IO Platform, compiled & flashed splitflap again and it works well now

I am pretty sure I missed something initially but still don't understand what exactly. Anyway issue is solved for me.

closed 2023-01-03 by manusovich

JoshuaPerk, 2023-12-02:
> 4\. Arduino TFT

> TTGO loads only once out of 10-20 times. For the rest it gives the following error:
>
> ```
> {"type":"init", "num_modules":5}
> E (10249) task_wdt: Task watchdog got triggered. The following tasks did not [...]

Getting the same error. Diagnosing.