scottbez1/splitflap #222: Loopback ERROR. Set output 0 but read incorrect value at input 0

brandonfake69, 2023-10-17:
I'm just setting one up and trying to Upload and Monitor. I get the following error:

```
{"type":"init", "num_modules":6}
{"msg": "Loopback ERROR. Set output 0 but read incorrect value at input 0", "type": "log"}
{"msg": "Loopback ERROR. Set output 1 but read incorrect value at input 1", "type": "log"}
f~�f~��x�x������x�x~`怘�怘���~�f�f�~f������f�f`ff�������`�~f�x�ff�f`�f�f`�����f����`f�������xf�~��f��~���f�ff~fxff�������fx�~���fx�~��`�fx�~�����fx�~����`�fx�~��~�fx�~�������`fx�`�������������x��x�~�������x��xx�xx�����f���怘�����x��xx�xx�������f�����`�������x������f�������x`�������x����f�����x����f��`��f���

{"type":"init", "num_modules":6}
{"msg": "Loopback ERROR. Set output 0 but read incorrect value at input 0", "type": "log"}
{"msg": "Loopback ERROR. Set output 1 but read incorrect value at input 1", "type": "log"}
```

I have 1 chainlink driver connected but and I'm not really sure where to go from here.

Thanks!

brandonfake69, 2023-10-17:
I have solid red LEDs on all 6 of the channels

scottbez1, 2023-10-17:
Most commonly this will occur with an incomplete crimped IDC connector, or with potential incomplete or bridged solder joints on the connectors on the PCBs. I would recommend starting there to see if anything stands out as a potential issue; if not, or if you're unsure, sending a high res photo of the whole setup (ESP32 and Chainlink driver), as well as any close-ups of solder joints you're unsure about can often be helpful to spot things.

closed 2023-10-27 by brandonfake69