scottbez1/splitflap #261: Bipolar Modification Support

ksbarnes, 2025-04-26:
I was potentially looking into the 28BYJ-48 Bipolar Mod (https://ardufocus.com/howto/28byj-48-bipolar-hw-mod) to have the motors produce more torque. I'm guessing the Chainlink boards wouldn't be able to support this out of the box (since the mod uses a different driver)? Assuming not, how difficult would it be to modify the chainlink boards?

scottbez1, 2025-04-26:
Switching to bipolar motors would require a pretty substantial rearchitecture of the Chainlink Drivers and firmware. Instead of the simple chained shift-registers with a single transistor per motor phase, you'd need 2 full H-bridge drivers for every motor (or a dedicated stepper motor driver for every motor). Personally I find the torque of good 12v 28byj-48 motors to be sufficient for a pretty reasonable and reliable splitflap rotation speed, so the effort of modifying every motor and the added expense of the driver electronics isn't something I've explored.

ksbarnes, 2025-04-26:
@scottbez1 thanks for the fast response; makes sense.

closed 2025-04-26 by ksbarnes